realsenseai/librealsense #5343: T265:Possible regression in behaviour of get_static_node()

neilyoung, 2019-11-28:
Hi,

I'm noticing a possible regression in behaviour of get_static_node. My current firmware version is 0.2.0.872, SDK 2.30.1, master branch

Maybe two weeks ago I could `get_static_node()` over several sessions with the T265 (USB not unplugged/replugged during runs), and I always got my marker back, w/o having to store/reload the map between runs. The marker was found within 30 s after start.

I also don't notice the RELOCATION event anymore in realsenseviewer, over several starts.

This seems to have changed. It seems, that the map does no longer return the marker after restart, if there is one set.

`get_static_marker()` works fine, if I load the map immediately at start. But as said: That wasn't always necessary. In fact, during my tests it was sometimes counterproductive to see the map relocating, so that I was forced to make the T265 forget the map by unplugging/replugging...

But now I miss that :)

Does that sound familiar?

neilyoung, 2019-11-28:
Same behaviour as described with firmware 0.2.0.857. Adapting now to firmware 0.2.0.879. I'm sure I was able to proceed on a map in memory between different sessions. I want that back :)

neilyoung, 2019-11-28:
Also not with 0.2.0.879 anymore. Very strange... Not a firmware issue?

giovannicairoli, 2019-12-03:
Hello @neilyoung,

Could you please help me sharing the importing-exporting map code? I really don't know how to implement the it.

Thank you very much,

Giovanni

neilyoung, 2019-12-03:
This is my experimental way, but you might find a more sophisticated approach in this PR

https://github.com/honpong/librealsense/blob/ar-advanced/examples/ar-advanced/rs-ar-advanced.cpp

I'm reading a map if provided in the args. I'm not sure, if my way is correct at all, since I need to start the pipe first in order to obtain a pose_sensor object, but I didn't spent much energy in optimizing this 

```
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    // Start pipeline with chosen configuration
    auto profile = pipe.start(cfg);
    auto device = profile.get_device();
    rs2::pose_sensor pose_sensor = device.first<rs2::pose_sensor>();

    if (argc == 2)
    {
        pipe.stop();

        printf("%s\r\n", argv[1]);
        std::vector<uint8_t> readfile;

        std::ifstream infile(argv[1], std::ios_base::binary);

        std::istreambuf_iterator<char> iter(infile);
        std::copy(iter, std::istreambuf_iterator<char>(), std::back_inserter(readfile)); // this leaves newVector empty

        infile.close(); // close  for consistency

        std::string str;
        str.assign(readfile.begin(), readfile.end());
        std::vector<uint8_t> vec(str.begin(), str.end());

        printf("%s, read size=%lu\r\n", argv[1], vec.size());
        if (pose_sensor.import_localization_map(vec))
        {
            printf("%s successfully imported\r\n", argv[1]);
        }
        else
        {
            printf("import failed, starting from scratch\r\n");
        }
        pipe.start(cfg);
    }
```

Then I'm having a kbhit/getch pattern, which allows me to non-blocking wait for inputs. On "Enter" I'm waiting a while and saving the map.

```
        if (kbhit())
        {
            int c = getch();

            if (c == 0x0d)
            {
                printf("stopping pose_sensor\r\n");
                pipe.stop();
                std::printf("exporting map ... sleeping 5s \r\n");
                std::this_thread::sleep_for(std::chrono::seconds(10));
                std::vector<uint8_t> map = pose_sensor.export_localization_map();
                printf("map exported, result: %lu bytes\r\n", map.size());
                printf("writing map to file\r\n");

                std::ofstream outfile("./map", std::ios::out | std::ofstream::binary);
                std::copy(map.begin(), map.end(), std::ostreambuf_iterator<char>(outfile));
                outfile.flush();
                outfile.close();
                break;
            }
       }

```

kbhit and getch are here:

```

#include <chrono>
#include <thread>
#include <iterator>
#include <iostream>
#include <fstream>
#include "boost/asio.hpp"

using namespace boost::asio;
struct termios orig_termios;

void reset_terminal_mode()
{
    tcsetattr(0, TCSANOW, &orig_termios);
}

void set_conio_terminal_mode()
{
    struct termios new_termios;

    /* take two copies - one for now, one for later */
    tcgetattr(0, &orig_termios);
    memcpy(&new_termios, &orig_termios, sizeof(new_termios));

    /* register cleanup handler, and set the new terminal mode */
    atexit(reset_terminal_mode);
    cfmakeraw(&new_termios);
    tcsetattr(0, TCSANOW, &new_termios);
}

int kbhit()
{
    struct timeval tv = {0L, 0L};
    fd_set fds;
    FD_ZERO(&fds);
    FD_SET(0, &fds);
    return select(1, &fds, NULL, NULL, &tv);
}

int getch()
{
    int r;
    unsigned char c;
    if ((r = read(0, &c, sizeof(c))) < 0)
    {
        return r;
    }
    else
    {
        return c;
    }
}

int kbhit()
{
    struct timeval tv = {0L, 0L};
    fd_set fds;
    FD_ZERO(&fds);
    FD_SET(0, &fds);
    return select(1, &fds, NULL, NULL, &tv);
}

int getch()
{
    int r;
    unsigned char c;
    if ((r = read(0, &c, sizeof(c))) < 0)
    {
        return r;
    }
    else
    {
        return c;
    }
}

```

You need to call  `set_conio_terminal_mode()` somewhere in the beginning of you "main" and find a way to reset the console behavior by  `reset_terminal_mode()`... Not sure, if this `asio` stuff is still required or just debris of previous tests. The new console behavior requires `\r\n` printfs.

neilyoung, 2019-12-03:
Optimized version appreciated :)

bfulkers-i, 2019-12-04:
@neilyoung your way to get a `pose_sensor` is a good one, here's a more minimal version:

```
#include <librealsense2/rs.hpp>
#include <fstream>

std::vector<uint8_t> load_file(const char * filename)
{
    std::ifstream stream(filename, std::ios::binary);
    std::vector<uint8_t> contents((std::istreambuf_iterator<char>(stream)), std::istreambuf_iterator<char>());
    return contents;
}

int main(int argc, char * argv[])
{
    rs2::log_to_console(RS2_LOG_SEVERITY_INFO);
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

    auto tm_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();
    auto map = load_file("map.map");
    // you can only import a map before the pipe has started
    tm_sensor.import_localization_map(map);

    pipe.start(cfg);
    // your program here
    pipe.stop();

    // currently, you can only export a map once the pipe has stopped
    std::vector<uint8_t> map_to_export = tm_sensor.export_localization_map();
    return 0;
}
```

neilyoung, 2019-12-04:
Thanks for sharing, Brian :) Is it still suggested to wait a couple of seconds before saving after stopping? I guess, if one can ignore the rest in the pipes, he can save right away after stopping, right?

bfulkers-i, 2019-12-04:
@neilyoung in my testing on #5213 there is no need to wait once you have called `pipe.stop()` and the call has completed

giovannicairoli, 2019-12-04:
Thank you both of you. I will try it as soon as possible.

giovannicairoli, 2019-12-04:
hello @neilyoung,

Just two super simple question for you,

I was trying the sample ar-advanced. I noticed that it can't import and export the map, probably because I didn't specified the path to the localization map.

The sample say to this:

Users should specify the path(s) to the localization map data file(s) by using command line arguments. For example, to set input map file src_map.raw at current directory:

`rs-ar-advanced --load_map src_map.raw`

To set both input map file src_map.raw and output map file dst_map.raw at current directory:

`rs-ar-advanced --load_map src_map.raw --save_map dst_map.raw`

Then, the application will import localization map data from src_map.raw at the beginning of tracking and will export the modified map including a virtual object as a static node to dst_map.raw.

Note: if neither input nor output path is given, this application will behave exactly the same as the rs-ar-basic example.

I new in programming and I don't know where add this code line:
`rs-ar-advanced --load_map src_map.raw --save_map dst_map.raw`

The other question is related to your code and @bfulkers-i, where do you specify the map path for import/export ?

Thank you for your help and patient!!

neilyoung, 2019-12-05:
@giovannicairoli Sorry for the late reply. 

```
I new in programming and I don't know where add this code line:
rs-ar-advanced --load_map src_map.raw --save_map dst_map.raw

```

These are not code lines, but bash command lines. They show how to specify in and out path/names for the maps on start of the rs-ar-advanced sample.

But I'm not the author of this example.

giovannicairoli, 2019-12-05:
@neilyoung hello,
Thank you for your answer. So also you are not able to run the sample and import export the map?

Thank you

neilyoung, 2019-12-05:
@giovannicairoli  No not yet. I'm a bit swamped with other things, sorry.

radfordi, 2019-12-05:
Thanks @neilyoung.  I think we have found the issue here.

neilyoung, 2019-12-06:
@radfordi I would say you should leave it as is now. This is really the best option to "OPT-IN" explicitly. 

Thanks. I will close this now.

neilyoung, 2019-12-06:
@radfordi Tested your sample code. It misses a "save_file" function. And for some reasons this sequence

```
pipe.stop();
std::vector<uint8_t> map_to_export = pose_sensor.export_localization_map();

```

ends up in 

```
RealSense error calling rs2_get_raw_data(buffer:nullptr):
    null pointer passed for argument "buffer"

```

Just in case you have something, don't waste your time.

neilyoung, 2019-12-10:
The current #5213 does not reliably save the map if you don't wait after pipe close. In most cases the response to an immediate save after pipe close on macOS is:

`RealSense error calling  rs2_get_raw_data (buffer:nullptr): null pointer passed for argument "buffer"
`

neilyoung, 2019-12-23:
I think this is fixed (not sure), will reopen in case

closed 2019-12-23 by neilyoung

giovannicairoli, 2020-01-20:
Hello @neilyoung,

Do you have any procedure/code about how initialize a map with the first static node"Origin", and then how to continue with set/get_static node procedure in order to save other static node positions with respect to the origin?

It is going to be a  bit tricky for me and there is no documentation about it.

Thank you very much,

Giovanni

neilyoung, 2020-01-20:
@giovannicairoli No. I don't think this is necessary. Right now I'm saving one static node, which is connected to some coordinate-conversion information. If the map is well trained you can go anywhere in the room/floor, the static node will be returned to you, if the cam detects, that a reprojection can happen ("having seen this already"). Then the returned static node will be returned in the your current camera reference frame.

Example (completely arbitrary) Start with your cam at your desk (0,0,0). Rotate 90 degrees right and move straight ahead 10 m. Store a static node. Walk around a while training. Save the map. 

Start a new attempt  from anywhere in the trained area with the previously created map loaded. Try to retrieve the static node while walking. Check the coordinates of the returned static node: If you have noted, at which place in the room the static node was placed, then the X/Z plane coordinates of the static node in the new camera reference frame should point exactly to that place in real world.

giovannicairoli, 2020-01-20:
Hello @neilyoung,

Yes your procedure seems to be very simple. So, if I understood.
If I don't have a map already. Firstly I create it and I write a set_static_node. How can I assign to it the coordinates(0,0,0)?. Because I tried to set some points and the coordinates were like: 1e8, 7e8 ...
Once placed the first static node I do the export map. 
After that running again the code I do the import_map every time and I wait until the camera get the static node previously saved? 
Then I can go around and set other static node continuously? and then export_map once finished?

How many and which distance you usually set a static node?

Thank you very much,

Giovanni

neilyoung, 2020-01-20:
We should establish a more direct contact. Feel free to share your email or any other contact with me. I'm looking for POC tests :)

neilyoung, 2020-01-20:
But to answer your question: One static node per map is sufficient.

giovannicairoli, 2020-01-20:
Hello @neilyoung,

Yes I need a bit a extra help, cairoli.giovanni@gmail.com.

Just one static node per map? Probably I have not understood the role of static node yet :(

Write to me in private by email please.

Have a nice day

giovanni

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi @neilyoung

Are you still having this issue in the latest librealsense 2.32.1?

neilyoung, 2020-02-03:
@RealSenseCustomerSupport No. As Jim stated here, it was indeed a kind of regression, since the map preservation - default before 2.32.1 - does now need an explicit enabling. Jim explained it here https://github.com/IntelRealSense/librealsense/issues/5343#issuecomment-562363468 and it was OK for me, so this issue has been closed and is fixed for me

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

giovannicairoli, 2020-04-07:
Hello,

from my side it can be close.

Thank you