realsenseai/librealsense #5345: How to reset the usb interface of camera with some error in python?

Dontla, 2019-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
I have six cameras connected to a win10 computer, but one of them is abnormal and it cannot be accessed normally. I want to reset it with python code, but I don’t know how to implement it with python statements. Can you give me some examples?
Best regard!
Dontla

MartyG-RealSense, 2019-11-28:
There was a RealSense user who adapted the **hardware_reset()** instruction for use with Python.

https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

The reply from #5428 should also apply to this ticket.
Closing this one unless otherwise. Thank you.
https://github.com/IntelRealSense/librealsense/issues/5428

closed 2019-12-13 by RealSenseCustomerSupport