realsenseai/librealsense #5347: How to use code ways to get an abnormal camera back to work?

Dontla, 2019-11-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
I use six cameras on a computer for target recognition, but occasionally, the camera connection is abnormal. I have tried many methods, such as hardware_reset() in pyrealsense, or use devcon tool to disable or enable or restart the USB device, but nothing help, the only solution is to unplug the USB cable and plug it in again, but doing so is not realistic for our device. I think if you guys can tell me a way to disconnect the power of USB by code ways(because I think it's probably equivalent to replugging), if so, please let me know.
Best regard!
Dontla

MartyG-RealSense, 2019-11-29:
If hardware_reset() does not work for you then you could try resetting the USB port with a script such as the one in the link below.

https://github.com/ralight/usb-reset/blob/master/usb-reset.c

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Please let us know if further assistance is needed. Thank you.

closed 2019-12-16 by RealSenseCustomerSupport