realsenseai/librealsense #5349: How long do we need to sleep after performing a camera hardware_reset?

Dontla, 2019-11-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
I use six cameras on a computer for target recognition, in order to ensure the operation of the camera, I added the hardware_reset() function at the beginning of the program, but some error will be reported during the operation, such as RuntimeError: Camera not connected! or RuntimeError: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult. So a time.sleep() function was added after the hardware_reset(), and the sleep time was 5 seconds. Most of the errors disappeared, but sometimes problems still occur, so I want to ask you how many seconds is the appropriate sleep delay time after the camera performs the hardware_reset()?
Best regard!
Dontla

MartyG-RealSense, 2019-11-30:
In another discussion about using sleep, @dorodnic suggested a large value such as 30 seconds, and the user later reported that errors no longer occurred when using this sleep period.

https://github.com/IntelRealSense/librealsense/issues/2866#issuecomment-446621127

dorodnic, 2019-12-01:
Re-enumeation is pretty fast for the camera itself, but it depends on the OS how fast it will register and propagate connection. On Windows in particular, the device driver information is stored in the Registry together with information about all devices that ever connected to that PC. This means that this delay is affected by other (non-RealSense) devices. 
One way to handle connection gracefully is to register to device callback on context. Another way is to sleep in a loop until new device is detected using query devices.

Dontla, 2019-12-02:
Hey @dorodnic ! I'm interested in the method of "One way to handle connection gracefully is to register to device callback on context", but I don't quite understand how to implement it, could you provide some code examples?
Best regard!
Dontla

Dontla, 2019-12-02:
Then I saw two methods in device.py of pyrealsense, but they are not commented. Do you know the usage of these two methods?
    def as_updatable(self): # real signature unknown; restored from __doc__
        """ as_updatable(self: pyrealsense2.pyrealsense2.device) -> rs2::updatable """
        pass
    def as_update_device(self): # real signature unknown; restored from __doc__
        """ as_update_device(self: pyrealsense2.pyrealsense2.device) -> rs2::update_device """
        pass
Best regard!
Dontla

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @Dontla

I hope you've been able to determine how to go about implementing the duration of sleep needed for your configuration.

Let us know if there is anything else further needed on this topic.

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport