realsenseai/librealsense #5350: How can I record audio using liberealsense SDK?

LeBW, 2019-11-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version | Win10  | 
| Platform                                 | PC. |
| SDK Version                            | 2.0                         | 
| Language                            |  {C++ }                          | 

### Issue Description
I want to develop an application to capture video(with audio). I know that SR300 has a microphone
so I bought it. But When I am in the process of developing, I don't find any API to control the microphone.( I use VS2015 and librealsense SDK2.0. ). How can I use the microphone to capture a video with audio?

MartyG-RealSense, 2019-11-30:
Yes, the cased USB versions of the SR300 and SR305 have a dual-array microphone.  I believe that the mic is treated like an ordinary USB webcam mic and so can be accessed as a device using an instruction called **uvc_get_libusb_handle**

http://docs.ros.org/kinetic/api/librealsense/html/group__device.html#ga74e368d1a8994fb8ea8bfe5d350899eb

This command may not be of use for a task as specific as recording audio from within a RealSense SDK 2.0 application though.  The standardized nature of the mic means that it should be accessible by other programs capable of recording audio, but probably not from within a Librealsense application.

One of the former lead developers of Librealsense called ddiakopoulos said about this subject in 2016:  "There is a separate USB audio interface that should be USB standards-compliant, so you should be able to access it using any cross-platform audio lib like RtAudio".

LeBW, 2019-11-30:
> Yes, the cased USB versions of the SR300 and SR305 have a dual-array microphone. I believe that the mic is treated like an ordinary USB webcam mic and so can be accessed as a device using an [...]

Thank you for explaining this. So there is no API to access the microphone in RealSense SDK, right? In fact, I think that in my application I don't need to use the deep image or something advanced, but just need to record a video like a ordinary camera(with audio, maybe MP4 format). So Maybe I don't need the Realsense SDK? Can you suggest me a framework to do this kind of thing with C++? Thank you.

MartyG-RealSense, 2019-11-30:
Windows has an API framework called DirectShow that has the ability to capture audio and video.

https://docs.microsoft.com/en-us/windows/win32/directshow/audio-capture

The DirectShow API is part of the 'Windows SDK', available from the link below.

https://developer.microsoft.com/en-us/windows/downloads/sdk-archive

DirectShow sample programs included with the Windows SDK are detailed here:

https://docs.microsoft.com/en-us/windows/win32/directshow/directshow-samples

dorodnic, 2019-12-01:
Hi @LeBW 
There is no APIs in the SDK for handling audio at the moment, and we did not put any audio recording devices in any of our products since the SR300. 
For the SR300, you should be able to use it as a normal web-cam on Windows / Linux, so I'd expect all the normal software to work as expected.

LeBW, 2019-12-01:
Than you all. Get it. :)

closed 2019-12-01 by LeBW