realsenseai/librealsense #5355: Unable to run ./scripts/patch-realsense-ubuntu-lts.sh

KR51K, 2019-12-02:
Hi,

   I try to install the realsense SDK on my Notebook (Ubuntu 18.04.3 LTS), but hanging at this point:

`./scripts/patch-realsense-ubuntu-lts.sh`

[....]

```
Patched kernels modules were created successfully

Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
Module is resident, unloading ... modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Failed to unload module hid_sensor_accel_3d. error type 1 . Operation is aborted
```

Hi tried this, but without any success:
`sudo modprobe -rf hid_sensor_accel_3d
modprobe: FATAL: Module hid_sensor_accel_3d is in use.
`

Any suggestions how to proceed?

Thanks and best regards!

ev-mp, 2019-12-02:
@KR51K , this may sometime occur regardless the camera connection state. The only I'm aware of to resolve it is to reboot with camera disconnected. The module `hid_sensor_accel_3d` is known to be prone to life-cycle management issues that we [circumvent in the SDK](https://github.com/IntelRealSense/librealsense/pull/3200), and  I suspect the error you receive is a related issue.

In several cases I observed the error popped when applying the patch with camera connected, so verifying that the camera is disconnected before running the script can alleviate this.

KR51K, 2019-12-03:
Hello,
thanks for your answer, but the camera is disconnected. Don't know but I want to install it on my Notebook - maybe this comes with internal sensors (convertable)?

ev-mp, 2019-12-03:
@KR51K , according to the original log the patched kernel modules were build successfully, and even though you weren't able to use them immediately they would become effective on reboot. 

Practically, in your case the installation was completed after reboot (like in the old good Windows times). Run `realsense-viewer` for verification and inspect the timestamp domain for the sensors - it should display `global/hw time` for all sensors including IMU.

A laptop should indeed have multiple sensors, but I haven't seen many with embedded accelerometer (unless it were a ruggedized MIL-STD one), so it is more likely that the error was generated after connecting and disconnecting IMU-enabled camera, like D435i.

KR51K, 2019-12-04:
Hello,
thanks. If tried it several times but the script stops after this error. I comment the line to get the gyro patched - without errors. 
But I think my main Problem is that the T265 is currently not detected correctly. When I use `lsusb` I get the result
`Bus 001 Device 005: ID 03e7:2150` 
It it even not working with realsense-viewer. So currently I don't know where to start finding the error - any suggestions?

   Thanks and best regards!

ev-mp, 2019-12-04:
@KR51K , the two things to check:
- `udev-rules` being properly installed
- When building from source make sure that `BUILD_WITH_TM2=true` cmake flag was set during the configuration stage. Note that the particular flag requires internet connection to download T265 image. Follow and inspect cmake log to verify that.

ev-mp, 2019-12-09:
@KR51K, do you need more assistance with the installation ?

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @KR51K,

Did you have a chance to check that udev-rules were properly installed and BUILD_WITH_TM2=true cmake flag was set during the configuration stage? Do you need any additional help with this matter?

Thank you!

closed 2019-12-23 by RealSenseCustomerSupport