realsenseai/librealsense #5359: UE4-Wrapper Editor crash in UE 4.22

mfruhner, 2019-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.30.0                          | 
| Language                            |Unreal Engine 4.22                         | 
| Segment			|  other                         | 

### Issue Description
Hi,
i followed the instructions on how to build an run the UE4-Wrapper project. I tried it with UE 4.22, because I need this version for another plugin. I can successfully run the TestLevel and see the camera output on the blocks.

When I try to open the Inspector Blueprint though, the Editor crashes every time with the following Stacktrace:
```

Unhandled exception

KERNELBASE
VCRUNTIME140
UE4Editor_RealSense!rs2::error::handle() [c:\program files (x86)\intel realsense sdk 2.0\include\librealsense2\hpp\rs_types.hpp:139]
UE4Editor_RealSense!FRealSenseInspectorCustomization::CustomizeDetails() [c:\users\maik\git\ue4-realsense\plugins\realsense\source\realsense\private\realsensedetailcustomization.cpp:157]
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_Kismet
UE4Editor_Kismet
UE4Editor_Kismet
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_Slate
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_Slate
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_Slate
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_Slate
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_SlateCore
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll
```

The docs say UE 4.19+ is supported, so I thought it should work. I haven't tried other versions yet.

Thanks for your feedback,
Maik

dorodnic, 2019-12-03:
Hi @mfruhner 
Thanks for the feedback
In general, we test with specific versions of UE and LRS, and fix issues with new versions based on user feedback
Please take a look at https://github.com/IntelRealSense/librealsense/pull/5306
I think it can address some of the issues

gaborpapp, 2019-12-04:
I tested this in Linux. If I open the inspector and start/stop several times I can make it crash after a while.

```
Application version 4.23.1.0
 ... built from changelist 0

OS version Linux 5.3.13-200.fc30.x86_64
Running 4 x86_64 processors (8 logical cores)
Exception was "SIGSEGV: invalid attempt to write memory at address 0x0000000000000003"

<CALLSTACK START>
libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [...UnrealEngine/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:413]
libUE4Editor-Core.so!FUnixErrorOutputDevice::Serialize(char16_t const*, ELogVerbosity::Type, FName const&) [...UnrealEngine/Engine/Source/Runtime/Core/Private/Unix/UnixErrorOutputDevice.cpp:70]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [...UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:71]
libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [...UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:101]
libUE4Editor-Core.so!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [...UnrealEngine/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:442]
libUE4Editor-OpenGLDrv.so!FOpenGLDynamicRHI::GetContextStateForCurrentContext(bool) [...UnrealEngine/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLDevice.cpp:165]
libUE4Editor-OpenGLDrv.so!UE4Function_Private::TFunctionRefCaller<FOpenGLDynamicRHI::RHIUpdateTexture2D(FRHITexture2D*, unsigned int, FUpdateTextureRegion2D const&, unsigned int, unsigned char const*)::$_69, void ()>::Call(void*) [...UnrealEngine/Engine/Source/Runtime/Core/Public/Templates/Function.h:486]
libUE4Editor-OpenGLDrv.so!FOpenGLDynamicRHI::RHIUpdateTexture2D(FRHITexture2D*, unsigned int, FUpdateTextureRegion2D const&, unsigned int, unsigned char const*) [...UnrealEngine/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLTexture.cpp:2304]
libUE4Editor-RealSense.so!FDynamicTexture::RenderCmd_UpdateTexture(FTextureUpdateData*) [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/DynamicTexture.cpp:185]
libUE4Editor-RealSense.so!FDynamicTexture::Update(rs2::video_frame const&) [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/DynamicTexture.cpp:125]
libUE4Editor-RealSense.so!ARealSenseInspector::ProcessFrameset(rs2::frameset*) [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/RealSenseInspector.cpp:385]
libUE4Editor-RealSense.so!ARealSenseInspector::WaitFrames() [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/RealSenseInspector.cpp:368]
libUE4Editor-RealSense.so!ARealSenseInspector::ThreadProc() [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/RealSenseInspector.cpp:305]
libUE4Editor-RealSense.so!FRealSenseInspectorWorker::Run() [...tools/librealsense/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Public/RealSenseInspector.h:189]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [...UnrealEngine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [...UnrealEngine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:177]
libpthread.so.0!UnknownFunction(0x84bf)
libc.so.6!clone(+0x42)

<CALLSTACK END>
```

mfruhner, 2019-12-11:
Hey,
I updated to the new version 2.31, which merged the Linux patches.
Now it doesn't build on Windows in Version 4.22 or 4.23 at all. Didn't test 4.24 yet.

> "librealsense2/rs.hpp": No such file or directory	RealSenseUE	C:\Users\Maik\git\ue4-realsense\Plugins\RealSense\Source\RealSense\Public\RealSenseNative.h	7	
> 
> 'UStaticMesh::SourceModels': Please do not access this member directly; use UStaticMesh::GetSourceModel(LOD) or UStaticMesh::GetSourceModels(). Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.	RealSenseUE	C:\Users\Maik\git\ue4-realsense\Plugins\RuntimeMeshComponent\Source\RuntimeMeshComponentEditor\Private\RuntimeMeshComponentDetails.cpp	490

dorodnic, 2019-12-11:
Missing `rs2.hpp` is just a compilation issue. 
I have executed `sudo make DESTDIR="/.../UnrealEngine-4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu" install` to make UE see `rs2.hpp` and `-lrealsense2` (tested with 4.24)
@gaborpapp tried it with 4.23 and 4.24 and mentioned he had to run `export LD_LIBRARY_PATH=/usr/local/lib64`

gaborpapp, 2019-12-11:
I haven't tested the changes on Windows, but there should not be any Linux specific about it. The missing header file can be caused by this change, where the Windows settings were moved:

https://github.com/IntelRealSense/librealsense/pull/5306/commits/fbe187c8c65c8cad18b1ca83192322025ff68527#diff-edd341da6ac45e3782b96ea2ff2c62b2R44

Can you test that the Windows conditional settings are executed?

mfruhner, 2019-12-11:
After updating to 4.23.1 it works again.
Seems to be the minimum version now?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@mfruhner Glad to see the issue resolved. Andy other questions about this? Thanks!

gaborpapp, 2019-12-13:
I tested it with 4.23.1 and 4.24, but it should not be too difficult to keep backwards compatibility with some #ifdef's if the changes break compilation in 4.22 and 4.23. I can try looking into this in some time if you need it.

closed 2020-01-07 by RealSenseCustomerSupport