realsenseai/librealsense #5362: Add post-processing filters in Unity

mintos002, 2019-12-03:
Hello!

I'm trying to add post-processing filter to the Depth image like the example for Unity: `PointCloudProcessingBlock`, but once the filter is applied I need to get the Color and Depth frames to manually process them. So I need a way to apply those filters without going throug the `RsProcessingPipe` to be able to get the post-processed frames in the same thread.

Here is my main function:

`void Update()
    {
        if (Input.GetKey("escape"))
        {
            UnityEngine.Debug.Log("ESC pressed.");
            Application.Quit();
        }

        if (!Streaming)
            return;

        if (processMode != ProcessMode.UnityThread)
            return;

        FrameSet frames;
        #region start_watch_timer
        Stopwatch watch = Stopwatch.StartNew();
        #endregion start_watch_timer
        if (m_pipeline.PollForFrames(out frames))
        {
            using (frames)
            using (Frame aligned_frames = align.Process(frames))
                if (aligned_frames.IsComposite)
                {
                    using (FrameSet fs = aligned_frames.As<FrameSet>())
                    using (VideoFrame color = fs.ColorFrame)
                    using (VideoFrame depth = fs.DepthFrame)
                    {
                        /// Insert code HERE for unity thread

                        counter++;
                        #region print_n_stop_timer
                        watch.Stop();
                        var curr = watch.ElapsedMilliseconds;
                        if (counter % 30 == 0)
                        {
                            //UnityEngine.Debug.Log("FPS = " + (1.0f / ((float)curr / 1000.0f)));
                        }
                        #endregion print_n_stop_timer

                        if (isSaveImages)
                        {
                            /// Write test
                            //cs2cpp.writeRGBimage(color.Data, "imageTest/Color_img_" + counter + ".png", color.Width, color.Height);
                            //cs2cpp.writeGRAYimage(depth.Data, "imageTest/Depth_img_" + counter + ".png", depth.Width, depth.Height);

                            /// RGB to BGR test
                            IntPtr color_bgr = cs2cpp.rgb2bgr(color.Data, color.Width, color.Height);
                            //cs2cpp.writeRGBimage(color_bgr, "imageTest/Color_img_bgr_" + counter + ".png", color.Width, color.Height);

                            /// Find circles test
                            Structures.image_struc struc_img;
                            //struc_img = cs2cpp.findCircles(color.Data, color.Width, color.Height);
                            struc_img = cs2cpp.findCircles(color_bgr, color.Width, color.Height);
                            
                            /// If circles where found
                            if (struc_img.vec_size != 0)
                            {
                                /// Print structure to txt file, show image of the structure
                                //cs2cpp.printStruct(struc_img, "imageTest/circles_" + counter + ".txt");
                                //cs2cpp.writeRGBimage(struc_img.data, "imageTest/Color_circles_" + counter + ".png", color.Width, color.Height);

                                /// Get the circles position from the structure
                                float[] c = new float[struc_img.vec_size];
                                System.Runtime.InteropServices.Marshal.Copy(struc_img.circles, c, 0, struc_img.vec_size);

                                /// Print the position of the circles
                                UnityEngine.Debug.Log("========================================");
                                for (int i = 0; i < struc_img.vec_size; i++)
                                {
                                    UnityEngine.Debug.Log(c[i]);
                                }

                                /// Initiate texture
                                if (texture == null || texture.width != color.Width || texture.height != color.Height)
                                {
                                    texture = new Texture2D(color.Width, color.Height, TextureFormat.RGB24, false);
                                }

                                /// Get canvas where texture is going to show
                                //GameObject rawImgObj = GameObject.Find("ResultImage");
                                //RawImage rawImg = rawImgObj.GetComponent<RawImage>();

                                /// Get circle color image in bgr
                                IntPtr img_cercles = cs2cpp.rgb2bgr(struc_img.data, struc_img.width, struc_img.height);

                                /// Insert texture to RawImage
                                texture.LoadRawTextureData(img_cercles, struc_img.width * struc_img.height * 3);
                                texture.Apply();
                                rawImg.texture = texture;

                                /// Free pointers
                                System.Runtime.InteropServices.Marshal.FreeHGlobal(img_cercles);

                                /// Clear Structure
                                struc_img.clear();
                            }
                            
                            /// Free pointers
                            System.Runtime.InteropServices.Marshal.FreeHGlobal(color_bgr);
                        }
                        /// Raise processed frames
                        RaiseSampleEvent(fs);
                    }
                }
            /// Raise original frames
            //RaiseSampleEvent(frames);
        }
    }`

I would like to do something like I did with `using(Frame aligned_frames = align.Process(frames))` but applying the post-processed filters in the `PointCloudProcessingBlock` example.

Is there any way I can apply directly to the frameset those filters? As long as I later can get the frames to modify them any answer will do!

Thanks!

wodndb, 2019-12-19:
I think there are no way to directly adjust filter for raw frames. Because filter is encapsulated. So I get aligned frames manually using custom RsStreamTextureRenderer.cs and RsPointCloudRenderer.cs

MartyG-RealSense, 2020-03-04:
@wodndb  thanks for your help in this case!

Hi @mintos002 - do you need any further assistance with this case or is it okay to close it please?  Thanks!

MartyG-RealSense, 2020-03-12:
Case closed due to no further responses.

closed 2020-03-12 by MartyG-RealSense