realsenseai/librealsense #5363: camera imu extrinsics calibration

vishnukumarts, 2019-12-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |  | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |

### Issue Description
I went through this document, 
https://www.intelrealsense.com/wp-content/uploads/2019/07/Intel_RealSense_Depth_D435i_IMU_Calibration.pdf

This document helps only with Intrinsic calibration i also require todo extrinsics(camera +imu) calibration.  Is there is any support document available?

schmidtp1, 2019-12-05:
Hi @vishnukumarts, you could use [kalibr](https://github.com/ethz-asl/kalibr) or [vicalib](https://github.com/arpg/vicalib) for example. (Please refer to their respective documentation.) Both of them support visual-inertial calibration.

ev-mp, 2019-12-09:
@vishnukumarts, as stated in the [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#imu-calibration) each D435i has a predefined IMU extrinsic data based on CAD measurements.

Additionally, the published IMU calibration script/procedure does not limit itself to the intrinsic scale and bias, but also rectifies the so-called `off-axis terms` - [𝑐𝑥𝑦, 𝑐𝑦𝑥, 𝑐𝑥𝑧, 𝑐𝑧𝑥, 𝑐𝑧𝑦, 𝑐𝑦𝑧] (Chapter 2.2 in the attached pdf) which imposes orthogonality on the accelerometer's axes and de-facto rectifies the sensor's rotational part of the extrinsic matrix. 
The translation part of the IMU extrinsic does not affects gyro readings and with regards of acceleration  - its effect is negligible given the sensor's dimensions and the fixed location where it can be physically attach to the PCB. 
So the calibration script handles the off-axis terms,  sensitive and offset parameters while the translation vector remains hard-coded based on the CAD measurements and cannot be modified.

closed 2019-12-17 by dorodnic