realsenseai/librealsense #5365: T265 Depth map in c++ / opencv?

antithing, 2019-12-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Operating System & Version |   {Win (10)   | 
                                     | 
| Platform                                 | PC |

| Language                            |  {C++ }                          | 

### Issue Description

I am trying to replicate the python T265 depth map example in c++, and I am getting a very noisy disparity map.

I get the intrinsics and extrinsics using the API, then convert the left and right frames to cv::Mat.

I then run this on both frames, passing an index of 1 for the left camera and 2 for the right.:

```
cv::Mat undistort(cv::Mat tmp_img, int index)
{
	cv::Mat InImage = tmp_img.clone();

	//undist	
	if (index == 1)
	{

		cv::Mat identity = cv::Mat::eye(3, 3, CV_64F);
		// Camera 1 Projection Matrix K[I|0]
		cv::Mat PP1(3, 4, cv::DataType<float>::type);
		camera_matrixL.copyTo(PP1.rowRange(0, 3).colRange(0, 3));

		//dist as 4
		cv::Mat distortionCoefficients = (cv::Mat1d(4, 1) << dist_coeffsL.at<double>(0, 0), dist_coeffsL.at<double>(1, 0), dist_coeffsL.at<double>(2, 0), dist_coeffsL.at<double>(3, 0));

		cv::Mat lmapx, lmapy, rmapx, rmapy;
		cv::fisheye::initUndistortRectifyMap(camera_matrixL, distortionCoefficients, identity, PP1, cv::Size(848, 800), CV_16SC2, lmapx, lmapy);
		cv::Mat undist_left;
		cv::remap(InImage, undist_left, lmapx, lmapy, cv::INTER_LINEAR);

		return undist_left;
	}
	if (index == 2)
	{
		cv::Mat identity = cv::Mat::eye(3, 3, CV_64F);
		cv::Mat PP1(3, 4, cv::DataType<float>::type);
		camera_matrixR.copyTo(PP1.rowRange(0, 3).colRange(0, 3));

		T.copyTo(PP1.rowRange(0, 3).col(3)); //extrinsic transformation between cams

		//dist as 4
		cv::Mat distortionCoefficients = (cv::Mat1d(4, 1) << dist_coeffsR.at<double>(0, 0), dist_coeffsR.at<double>(1, 0), dist_coeffsR.at<double>(2, 0), dist_coeffsR.at<double>(3, 0));

		cv::Mat lmapx, lmapy, rmapx, rmapy;
		cv::fisheye::initUndistortRectifyMap(camera_matrixR, distortionCoefficients, identity, PP1, cv::Size(848, 800), CV_16SC2, lmapx, lmapy);
		cv::Mat undist_right;
		cv::remap(InImage, undist_right, lmapx, lmapy, cv::INTER_LINEAR);

		return undist_right;
	}

}

```

My stereo code is:

```

                	cv::Mat undL = undistort(frameLeft, 1);
			cv::Mat undR = undistort(frameRight, 2);
		
			//depth
			cv::Mat disparity_left = cv::Mat(frameLeft.rows, frameLeft.cols, CV_16S);
			cv::Mat g1, g2, disp, disp8;

			g2 = frameLeft.clone();
			g1 = frameRight.clone();
			cv::Ptr<cv::StereoBM> sbm = cv::StereoBM::create(0, 21);
			sbm->setDisp12MaxDiff(1);
			sbm->setSpeckleRange(8);
			sbm->setSpeckleWindowSize(9);
			sbm->setUniquenessRatio(0);
			sbm->setTextureThreshold(507);
			sbm->setMinDisparity(-39);
			sbm->setPreFilterCap(61);
			sbm->setPreFilterSize(5);
			sbm->compute(g1, g2, disparity_left);
			cv::normalize(disparity_left, disp8, 0, 255, cv::NORM_MINMAX, CV_8U);

			cv::imshow("disp", disp8);
			cv::waitKey(1);
```

What am I doing wrong?

Thanks!

schmidtp1, 2019-12-04:
@bfulkers-i

antithing, 2019-12-04:
I have moved on to trying to convert the Python from the example to c++. I have the following, but it gives me black images. 

```
void RSUNDIST(cv::Mat left, cv::Mat right)
{
	double vis_mult = 5.0;
	int wsize = 5;
	float min_disp = 0;
	int num_disp = 112 - min_disp;
	int max_disp = min_disp + num_disp;
	double lambda = 10000.0;
	double sigma = 1.0;

	cv::Ptr<cv::StereoSGBM> left_matcher = cv::StereoSGBM::create(0,
		num_disp,
		16,
		8 * 3 * wsize*wsize,
		32 * 3 * wsize*wsize,
		1,
		10,
		100,
		32);
	
	left_matcher->setMode(cv::StereoSGBM::MODE_SGBM);

	    float stereo_fov_rad = 90 * (M_PI / 180); // 90 degree desired fov
		float stereo_height_px = 300;          // 300x300 pixel stereo output
		float stereo_focal_px = stereo_height_px / 2 / tan(stereo_fov_rad / 2);

		// We set the left rotation to identity and the right rotation
		// the rotation between the cameras
		cv::Mat R_left = cv::Mat::eye(3, 3, CV_32F);
		cv::Mat R_right = R;

		// The stereo algorithm needs max_disp extra pixels in order to produce valid
		// disparity on the desired output region.This changes the width, but the
		// center of projection should be on the center of the cropped image
		float stereo_width_px = stereo_height_px + max_disp;
		cv::Size stereo_size = cv::Size(stereo_width_px, stereo_height_px);
		float stereo_cx = (stereo_height_px - 1) / 2 + max_disp;
		float stereo_cy = (stereo_height_px - 1) / 2;

		// Construct the left and right projection matrices, the only difference is
		// that the right projection matrix should have a shift along the x axis of
		// baseline*focal_length

		cv::Mat_<float> P_left(3, 4); // rows, cols
		P_left << stereo_focal_px, 0, stereo_cx, 0,
		0, stereo_focal_px, stereo_cy, 0,
	    0, 0, 1, 0;

	

		cv::Mat P_right = P_left.clone();
		P_right.at<float>(0, 3) = T.at<float>(0, 0) * stereo_focal_px;

		// Construct Q for use with cv2.reprojectImageTo3D.Subtract max_disp from x
		// since we will crop the disparity later

		cv::Mat_<float> Q(4, 4); 
		P_left << 1, 0, 0, -(stereo_cx - max_disp),
		0, 1, 0, -stereo_cy,
			0, 0, 0, stereo_focal_px,
			0, 0, -1 / T.at<float>(0, 0), 0;

		cv::Mat lm1, lm2, rm1, rm2;
		cv::fisheye::initUndistortRectifyMap(camera_matrixL, dist_coeffsL, R_left, P_left, stereo_size, CV_32FC1, lm1, lm2);
		cv::fisheye::initUndistortRectifyMap(camera_matrixL, dist_coeffsL, R_left, P_left, stereo_size, CV_32FC1, rm1, rm2);

		cv::Mat undist_left, undist_right;
		cv::remap(left, undist_left, lm1, lm2, cv::INTER_LINEAR);
		cv::remap(right, undist_right, rm1, rm2, cv::INTER_LINEAR);

		cv::imshow("ll", undist_left);
		cv::imshow("rr", undist_right);

		cv::waitKey(1);

}
```

antithing, 2019-12-05:
By running the python example i was able to see what i was missing.

Creating the Projection matrices like this helped me get a pretty good result.

```
 //LEFT
		float fx = camera_matrixL.at<float>(0, 0);
		float fy = camera_matrixL.at<float>(1, 1);

		//Set mid point to actual pixel center
		cv::Mat_<float> P_left(3, 4);
		P_left << fx, 0, 424, 0,
			0, fy, 400, 0,
			0, 0, 1, 0;

//RIGHT

                float fx = camera_matrixR.at<float>(0, 0);
		float fy = camera_matrixR.at<float>(1, 1);

		//P2 gets the baseline added
		cv::Mat_<float> P_right(3, 4);
		P_right << fx, 0, 424, T.at<float>(0, 0) * 10,
			0, fy, 400, 0,
			0, 0, 1, 0;
```

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @antithing

Is there anything else needed on this thread?

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport

tiralonghipol, 2020-09-06:
hi @antithing 
do you maybe have a complete working example of this?
I am trying to achieve the same thing
Thank you!

bamboosdu, 2021-01-30:
Hi, @antithing 
I am trying to achieve the same thing. Can you share complete example?
Thank you!

tiralonghipol, 2021-01-30:
https://github.com/tiralonghipol/t265_depth