realsenseai/librealsense #5366: cannot get metadata to work on d435i with any kernel

nikisalli, 2019-12-03:
Camera model:     d435i
Firmware version: 5.10.13.00
os:                         ubuntu 18.04
kernel:                   tried 4.15.0 / 4.15.18 / 4.16 / 4.18 / 5.0.0-37 / 5.1.6
platform:                lattepanda alpha d864
sdk version:           latest, 1.3.10
environment:          ROS melodic

I cannot get the kernel patches to work. I tried everything I could find related to the topic on every kernel:
- [installing from apt](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
- [building from source](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)
- [building with libuvc](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)

the output of the command `modinfo uvcvideo | grep "version:"` keeps being:
`version:        1.1.1`
`srcversion:     4FA7FAEF8FFC0A9AB1BAC31`
and this causes ros to not find metadata and crash:
`UVC metadata payloads not available. Please refer to installation chapter for details. [ WARN] [1531210273.286442270]: Frame metadata isn't available!`

here is the output of `dmesg | grep uvcvideo` when i connect the camera:
[ 3168.313030] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 3168.315811] uvcvideo: Unable to create debugfs 2-2 directory.
[ 3168.315962] uvcvideo 2-4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 3168.315965] uvcvideo 2-4:1.0: Entity type for entity Processing 2 was not initialized!
[ 3168.315967] uvcvideo 2-4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 3168.315969] uvcvideo 2-4:1.0: Entity type for entity Camera 1 was not initialized!
[ 3168.316486] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 3168.318147] uvcvideo: Unable to create debugfs 2-2 directory.
[ 3168.318222] uvcvideo 2-4:1.3: Entity type for entity Processing 7 was not initialized!
[ 3168.318224] uvcvideo 2-4:1.3: Entity type for entity Extension 8 was not initialized!
[ 3168.318226] uvcvideo 2-4:1.3: Entity type for entity Camera 6 was not initialized!

and here is `cat /var/lib/dkms/librealsense2-dkms/1.3.10/build/make.log` on 5.0.0-37 kernel:
DKMS make.log for librealsense2-dkms-1.3.10 for kernel 5.0.0-37-generic (x86_64)
mar  3 dic 2019, 16.34.20, CET
make: Entering directory '/usr/src/linux-headers-5.0.0-37-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
scripts/Makefile.build:42: /var/lib/dkms/librealsense2-dkms/1.3.10/build/5.0.0/drivers/media/usb/uvc/Makefile: No such file or directory
make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.10/build/5.0.0/drivers/media/usb/uvc/Makefile'.  Stop.
Makefile:1786: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-5.0.0-37-generic'

output of `uname -r`:
`5.0.0-37-generic`

error line when installing librealsense2-dkms:
`sudo apt-get install --reinstall librealsense2-dkms`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/5.992 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 321636 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.10-0ubuntu1_all.deb ...

Deleting module version: 1.3.10
completely from the DKMS tree.

Done.
Unpacking librealsense2-dkms (1.3.10-0ubuntu1) over (1.3.10-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Current status:
mod: videodev 
mod: uvcvideo                    version: 1.1.1
mod: hid_sensor_gyro_3d 
mod: hid_sensor_accel_3d 
Setting up librealsense2-dkms (1.3.10-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.10 DKMS files...
Building for 5.0.0-37-generic
Building initial module for 5.0.0-37-generic
Error! Bad return status for module build on kernel: 5.0.0-37-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.10/build/make.log for more information.

Loading the modified modules into kernel...modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Current status:
librealsense2-dkms, 1.3.10: added
mod: videodev 
mod: uvcvideo                    version: 1.1.1
mod: hid_sensor_gyro_3d 
mod: hid_sensor_accel_3d

ev-mp, 2019-12-03:
@nikisalli hello, the attached logs is for kernel 5.0 while dkms package is intended for [Ubuntu Bionic LTS kernels 4.15 and 4.18](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages).

Can you clarify whether the other kernels that you mention -  4.15.0 / 4.15.18 / ~~4.16~~ / 4.18  were LTS kernels, as Ubuntu LTS numbers follow the **x.y.0-z** pattern, e.g. 5.0.0-37  (`uname -r`). 
In case of failure with the LTS branches 4.15 and 4.18 please post dkms installation log for review.

nikisalli, 2019-12-05:
ok my bad, I reinstalled everything on a clean ubuntu installation with 4.15-72 kernel and it works! thank you for pointing out the problem

closed 2019-12-05 by nikisalli