realsenseai/librealsense #5368: Change config parameters with rs-record tool

collinabidi, 2019-12-03:
|----------------------------------------|------------------------------------------- |
| Camera Model | D400 |
| Firmware Version | 2.29.0 |
| Operating System & Version | Ubuntu 18.04 |
| Kernel Version (Linux Only) | 5.0.0-31 |
| Platform | PC |
| SDK Version | 2.0 |
| Language | C++/Python |
| Segment | VR/AR |
|----------------------------------------|------------------------------------------- |

**Issue Description**

I'm modifying the C++ ```rs-record``` tool to attempt to record an RGB-D stream to a .bag file. I'm attempting to record 16-bit depth frames as well as 16-bit RGB frames, but it appears that the ```cfg.enable_record_to_file( ... )``` method doesn't have any way to specify parameters.

Is there a way to record to a .bag file with specified parameters?

ev-mp, 2019-12-03:
@collinabidi hi,
The recorder automatically picks all the streams available.
To customize the streaming configuration use `rs2::config` APIs:

```c++
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
cfg.enable_record_to_file("abc.bag");

pipeline.start(cfg)
```

Note that D430 doesn't provide RGB stream, only Depth and IR1/2

collinabidi, 2019-12-06:
Apologies, my camera is the D435.

That seemed to do the trick! Is there a way to change laser power or enable postprocessing steps?

RealSenseCustomerSupport, 2020-01-10:
----------------------------------------------

Anything modifiable is exposed via the rs2::config APIs.

Will you be needing any further assistance on this?

closed 2020-01-20 by RealSenseCustomerSupport