realsenseai/librealsense #5371: Set best config parameters to get depth map for face anti-spoofing

xiao-keeplearning, 2019-12-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 2.22.0 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17)  | 
| Platform                                 | PC  |
| SDK Version                            |  {2.0 }                          | 
| Language                            |  {C++/python }                          | 
| Segment			|  {face recognition and face anti-spoofing }                          | 

### Issue Description

> 0 - Jet
> 1 - Classic 
> 2 - WhiteToBlack 
> 3 - BlackToWhite
> 4 - Bio
> 5 - Cold
> 6 - Warm
> 7 - Quantized
> 8 - Pattern

What is the difference between these different rendering methods, and what kind of colorizer should I use in which scene? 
Is there documentation on this aspect?
I require to get depth map with relatively high depth accuracy within 0.3-1.5m and at the same time make the depth value and the color value(not change dynamically) one to one correspondence.

Here is my config:
```
rs2::colorizer _colorizer;
# set near mode
_colorizer.set_option(RS2_OPTION_VISUAL_PRESET, 2.f);
_colorizer.set_option(RS2_OPTION_COLOR_SCHEME, 0.f);
_colorizer.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 0);
_colorizer.set_option(RS2_OPTION_MIN_DISTANCE, 0.3f);
_colorizer.set_option(RS2_OPTION_MAX_DISTANCE, 2.f);
```
Actually, I also try to use WhiteToBlack randering method, but I found pixel values ​​are difficult to reflect relative depth. Please help me ?

xiao-keeplearning, 2019-12-04:
#https://github.com/IntelRealSense/librealsense/issues/2133#issuecomment-407519929

xiao-keeplearning, 2019-12-04:
https://github.com/IntelRealSense/librealsense/issues/2133#issuecomment-407519929

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@xiao-keeplearning The rendering method is for depth visualization in preview. It will not affect the depth value or accuracy. If you want to high depth accuracy, you can try preset "high accuracy".

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@xiao-keeplearning Any other questions for this? Looking forward to your update. Thanks!

xiao-keeplearning, 2019-12-23:
> @xiao-keeplearning The rendering method is for depth visualization in preview. It will not affect the depth value or accuracy. If you want to high depth accuracy, you can try preset "high accuracy".
Recently I tried “high accuracy” mode, I found some problem.

[image]
[image]
The first image is from the default mode，and the second image is from high accuracy mode. Why the second image has more the Invalid Depth Band?

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

@xiao-keeplearning This is what is expected with high accuracy preset. High accuracy preset is for high confidence threshold value of depth and lower fill factor. For some usage such as collision avoidance, the high confidence valid depth is more important than good fill factor. You can select the preset according to your usage. For more information about preset, please refer to https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@xiao-keeplearning Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@xiao-keeplearning Any update from your side? If we don’t hear from you in 7 days, this issue will be closed. Thanks!

xiao-keeplearning, 2020-02-14:
Thanks for your help，I have solved the problem temporarily.

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@xiao-keeplearning Glad to hear the issue resolved. Thanks for the update!

closed 2020-02-17 by RealSenseCustomerSupport