realsenseai/librealsense #5372: Announcement:. Intel white paper on Visual SLAM for RealSense T265 Tracking Camera

MartyG-RealSense, 2019-12-04:
Hi everyone,

Intel's archive of white paper' guide documents lists a new paper about using Visual SLAM with the RealSense T265.  

https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera 

Intel's full range of white papers can be viewed here:

https://dev.intelrealsense.com/docs/whitepapers

neilyoung, 2019-12-04:
Thanks, very interesting

neilyoung, 2019-12-09:
https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera

>Fig. 6. Example of tracking of April Tags. Screenshot reproduced from Video in Ref 4. April Tag tracking is included in the T265.

This is not entirely true, IMHO. The April Tag tracking is not included in the T265.

MartyG-RealSense, 2019-12-09:
I will re-open this discussion so that the Intel RealSense team members can see it and respond.  Thanks for the feedback!

dorodnic, 2019-12-10:
@bfulkers-i @agrunnet  - phrasing of the white-paper may be a bit misleading. I suggest "integrated with" instead of "supported by"

neilyoung, 2019-12-10:
No, no. That wasn't meant so :) Include it, and the text can stay as is :))

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

Trying to follow what is not true about the statement.  Is it the fact that AprilTag tracking is just a sample from a SW perspective and not fully integrated in the HW/FW of T265?  Is it that the statement that is in the whitepaper makes it to believe that the AprilTag tracking is on by default without any other work necessary on developer side?

neilyoung, 2019-12-25:
It is the fact, that the statement declares an integration. In fact, April Tag is an external application of the T265. With that in mind the answer to your question is yes

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Makes sense what is required.  Thanks.

closed 2020-01-02 by MartyG-RealSense