realsenseai/librealsense #5373: C# equivalent of some Python API

HEEmma, 2019-12-04:
I want to create a C# UI which needs the function of some Python API.
I wonder if there's C# equivalent of rs2_deproject_pixel_to_point, rs2_transform_point_to_point, rs2_project_point_to_pixel ?

MartyG-RealSense, 2020-03-04:
Apologies for the delay in responding to this case.

I could not find any support for the projection / deprojection functions in the C# API.  I did locate some links to RealSense users who had programmed workarounds to achieve the same goal though.

https://github.com/IntelRealSense/librealsense/issues/2657

https://pterneas.com/2018/10/10/intel-realsense-coordinate-mapping/

MartyG-RealSense, 2020-03-12:
This case will be closed after 7 days from now if there are no further questions.   Thanks!

closed 2020-03-19 by MartyG-RealSense