realsenseai/librealsense #5374: "There are 0 connected RealSense devices" message  on Jetson Nano

paul3747, 2019-12-04:
OS: Ubuntu 18.04 
Device : Realsense R200 camera 
Jetson Nano

I gitclone the library from https://github.com/IntelRealSense/librealsense/tree/v1.12.1#installation-guide
I followed instructions given on
https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                  | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  v1.12.1 }                          | 

### Issue Description
I have followed the installation-guide and finish make install，
But when I want to executethe example in /usr/local/bin ，
I tried the command "sudo ./rs-color", and  it return the error
"There are 0 connected RealSense devices"

paul3747, 2019-12-04:
And this is the result of dmesg -wT
[三 十二  4 17:25:19 2019] usb 1-2.1: USB disconnect, device number 3
[三 十二  4 17:25:19 2019] wlan0: deauthenticating from c4:e9:84:66:39:8e by local choice (Reason: 3=DEAUTH_LEAVING)
[三 十二  4 17:25:19 2019] rtl_usb: reg 0x102, usbctrl_vendorreq TimeOut! status:0xffffffed value=0x90e1317
[三 十二  4 17:25:19 2019] rtl_usb: reg 0x422, usbctrl_vendorreq TimeOut! status:0xffffffed value=0xf00069ce
[三 十二  4 17:25:19 2019] rtl_usb: reg 0x542, usbctrl_vendorreq TimeOut! status:0xffffffed value=0x6954342b
[三 十二  4 17:25:19 2019] rtl_usb: reg 0x1cc, usbctrl_vendorreq TimeOut! status:0xffffffed value=0x6954342b
[三 十二  4 17:25:22 2019] usb 1-2.3: new high-speed USB device number 5 using tegra-xusb
[三 十二  4 17:25:22 2019] usb 1-2.3: New USB device found, idVendor=7392, idProduct=7811
[三 十二  4 17:25:22 2019] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[三 十二  4 17:25:22 2019] usb 1-2.3: Product: 802.11n WLAN Adapter
[三 十二  4 17:25:22 2019] usb 1-2.3: Manufacturer: Realtek
[三 十二  4 17:25:22 2019] usb 1-2.3: SerialNumber: 00e04c000001
[三 十二  4 17:25:22 2019] rtl8192cu: Chip version 0x10
[三 十二  4 17:25:22 2019] rtl8192cu: Board Type 0
[三 十二  4 17:25:22 2019] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[三 十二  4 17:25:22 2019] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[三 十二  4 17:25:22 2019] ieee80211 phy1: Selected rate control algorithm 'rtl_rc'
[三 十二  4 17:25:22 2019] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[三 十二  4 17:25:22 2019] rtl8192cu: MAC auto ON okay!
[三 十二  4 17:25:22 2019] rtl8192cu: Tx queue select: 0x05
[三 十二  4 17:25:23 2019] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[三 十二  4 17:25:23 2019] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[三 十二  4 17:25:24 2019] wlan0: authenticate with c4:e9:84:66:39:8e
[三 十二  4 17:25:24 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:24 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:24 2019] wlan0: authenticated
[三 十二  4 17:25:24 2019] wlan0: associate with c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:24 2019] wlan0: associate with c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:24 2019] wlan0: associate with c4:e9:84:66:39:8e (try 3/3)
[三 十二  4 17:25:24 2019] wlan0: association with c4:e9:84:66:39:8e timed out
[三 十二  4 17:25:25 2019] wlan0: authenticate with c4:e9:84:66:39:8e
[三 十二  4 17:25:25 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:25 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:25 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 3/3)
[三 十二  4 17:25:26 2019] wlan0: authentication with c4:e9:84:66:39:8e timed out
[三 十二  4 17:25:27 2019] wlan0: authenticate with c4:e9:84:66:39:8e
[三 十二  4 17:25:27 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:27 2019] wlan0: authenticated
[三 十二  4 17:25:27 2019] wlan0: associate with c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:27 2019] wlan0: associate with c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:27 2019] wlan0: associate with c4:e9:84:66:39:8e (try 3/3)
[三 十二  4 17:25:27 2019] wlan0: association with c4:e9:84:66:39:8e timed out
[三 十二  4 17:25:29 2019] wlan0: authenticate with c4:e9:84:66:39:8e
[三 十二  4 17:25:29 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:29 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:30 2019] wlan0: send auth to c4:e9:84:66:39:8e (try 3/3)
[三 十二  4 17:25:30 2019] wlan0: authenticated
[三 十二  4 17:25:30 2019] wlan0: associate with c4:e9:84:66:39:8e (try 1/3)
[三 十二  4 17:25:30 2019] wlan0: associate with c4:e9:84:66:39:8e (try 2/3)
[三 十二  4 17:25:30 2019] wlan0: RX AssocResp from c4:e9:84:66:39:8e (capab=0x411 status=0 aid=3)
[三 十二  4 17:25:30 2019] wlan0: associated
[三 十二  4 17:25:30 2019] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[三 十二  4 17:25:31 2019] IPv6: wlan0: IPv6 duplicate address fe80::d461:7e34:84da:d745 detected!
[三 十二  4 17:25:35 2019] usb usb2: usb_suspend_both: status -16
[三 十二  4 17:25:35 2019] usb usb2: usb_suspend_both: status 0
[三 十二  4 17:25:35 2019] usb 2-1.1: new SuperSpeed USB device number 5 using tegra-xusb
[三 十二  4 17:25:35 2019] usb 2-1.1: New USB device found, idVendor=8086, idProduct=0a80
[三 十二  4 17:25:35 2019] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[三 十二  4 17:25:35 2019] usb 2-1.1: Product: Intel RealSense 3D Camera R200
[三 十二  4 17:25:35 2019] usb 2-1.1: Manufacturer: Intel Corp
[三 十二  4 17:25:35 2019] usb 2-1.1: SerialNumber: SN_2511015056
[三 十二  4 17:25:38 2019] nvgpu: 57000000.gpu             railgate_enable_store:343  [INFO]  railgate is disabled.
[三 十二  4 17:26:44 2019] IPv6: wlan0: IPv6 duplicate address fd24:a43a:d4b:5:f15f:5e6e:ca22:a077 detected!
[三 十二  4 17:26:44 2019] IPv6: ipv6_create_tempaddr: retry temporary address regeneration
[三 十二  4 17:26:44 2019] IPv6: wlan0: IPv6 duplicate address fd24:a43a:d4b:5:e59e:858c:616a:294e detected!
[三 十二  4 17:26:44 2019] IPv6: ipv6_create_tempaddr: retry temporary address regeneration
[三 十二  4 17:26:44 2019] IPv6: wlan0: IPv6 duplicate address fd24:a43a:d4b:5:bc9f:2739:fc1:94be detected!
[三 十二  4 17:26:44 2019] IPv6: ipv6_create_tempaddr: regeneration time exceeded - disabled temporary address support
[三 十二  4 17:26:44 2019] IPv6: wlan0: IPv6 duplicate address fd24:a43a:d4b:5:a8a7:2fd6:f6c:7e01 detected!
[三 十二  4 17:26:44 2019] IPv6: ipv6_create_tempaddr: use_tempaddr is disabled
[... cut ...]
[三 十二  4 17:29:39 2019] nvgpu: 57000000.gpu             railgate_enable_store:343  [INFO]  railgate is disabled.

And lsusb result
Bus 002 Device 005: ID 8086:0a80 Intel Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dorodnic, 2019-12-04:
Hi @paul3747 
There is no rs-color in 1.12.1 version, but you should try cpp-capture instead.

paul3747, 2019-12-04:
@dorodnic Thanks for your reply!
I have already tried rs-capture too
But it still cannot successfully execute, 
Below shows the error:
"
RealSense error calling rs2_pipeline_start(pipe:0x5584c324d0):
    No device connected
"

paul3747, 2019-12-04:
@dorodnic  
In addition, I have three devices connect to Jetson Nano by USB3.0,
One is my bluetooth keyboard and mouse sets, one is wifi module , another is R200,
When I execute the program related to the R200,
I cannot type any word by my keyboard,unless I remove the R200.
Do you know whats going on and how to solve it?

dorodnic, 2019-12-04:
In librealsense 2.x+ R200 support was deprecated, but version 1.12.1 should work with the R200. 
I suspect you are building 2.x version because 1.12.1 does not include `rs-color` or `rs-capture` but calls the same sample `cpp-capture`

paul3747, 2019-12-05:
@dorodnic Thanks for your recommendation!
I have already installed the right version SDK，
But beacuse I tried SDK1.12.1 and SDK2 both before，
So i ran the wrong example，
Now I tried the example "cpp-capture" of SDK1.12.1，
And it return this error:
"RealSense error calling rs_create_context(api_version:11201):
    uvcvideo kernel module is not loaded"
Have you meet this error before?

dorodnic, 2019-12-05:
What do you get from `sudo modprobe uvcvideo`?

paul3747, 2019-12-06:
I got
"modprobe ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)"

dorodnic, 2019-12-10:
Unfortunately, its a limitation of librealsense1 of not being able to work with monolithic kernels (where uvcvideo is not built as a loadable module). First generation of RealSense products (F200, R200, ZR300) were never designed to work on Jetson and other modern platforms. Its is not impossible (building the Kernel with uvcvideo as a module will overcome this specific problem) but is less streamlined compared to current generation.

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

@paul3747 Any other questions for this with the reply from dorodnic? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

@paul3747 Will close the issue accordingly if no further questions. Thanks!

closed 2020-01-07 by RealSenseCustomerSupport

paul3747, 2020-01-20:
@dorodnic @RealSenseCustomerSupport 
Dear all:
Sorry for late reply，
I have successfully overcome this problem by reloading the uvc_video module，
First, I thought I need to rebuild the kernel，and I learned how to do it for a long time,
but, after these works, I found the driver module have been existed in the /driver/media/usb/uvc/uvc_video.ko.backup
After I import this module by 
"insmod lib/modules/4.9.140-tegra/drivers/media/usb/uvc/uvc_video.ko"
and it works.

sebastienvg, 2020-06-12:
> @dorodnic @RealSenseCustomerSupport
> Dear all:
> Sorry for late reply，
> I have successfully overcome this problem by reloading the uvc_video module，
> First, I thought I need to rebuild the kernel，and I learned how to do it for a long time,
> but, after these works, I found the driver module have been existed in the /driver/media/usb/uvc/uvc_video.ko
> After I import this module by
> "insmod lib/modules/4.9.140-tegra/drivers/media/usb/uvc/uvc_video.ko"
> and it works.

I am very much so in the same situation but couldn't get it to work, I am running the following config

 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.4 DP [L4T 32.4.2]
   * NV Power Mode: MAXN - Type: 0
   * jetson_clocks service: active
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.145
   * TensorRT: 7.1.0.16
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.2.0
   * Vulkan: 1.2.70

And I've been trying to about 2 days now to get librealsense (legacy) running with R200.
I tried to unload and reload the module /lib/modules/4.9.140-tegra/kernel/drivers/media/usb/uvc/uvc_video.ko
but no go (I noticed you noted `lib/modules/4.9.140-tegra/kernel/drivers/media/usb/uvc/uvc_video.ko` without a / at the beginning and a _ in uvc_video.ko, I wasn't able to find those)

Upon reloading I get this in dmes

`[ 5385.986875] uvcvideo: Found UVC 1.00 device <unnamed> (046d:0825)
[ 5386.001106] uvcvideo 1-2.2:1.0: Entity type for entity Extension 4 was not initialized!
[ 5386.009217] uvcvideo 1-2.2:1.0: Entity type for entity Extension 6 was not initialized!
[ 5386.017282] uvcvideo 1-2.2:1.0: Entity type for entity Extension 7 was not initialized!
[ 5386.025307] uvcvideo 1-2.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 5386.033412] uvcvideo 1-2.2:1.0: Entity type for entity Extension 3 was not initialized!
[ 5386.041431] uvcvideo 1-2.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 5386.049443] input: UVC Camera (046d:0825) as /devices/70090000.xusb/usb1/1-2/1-2.2/1-2.2:1.0/input/input10
[ 5386.123213] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 5386.124830] uvcvideo 2-1.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 5386.132877] uvcvideo 2-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5386.140917] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 5386.142202] uvcvideo: Unable to create debugfs 2-6 directory.
[ 5386.142378] uvcvideo 2-1.1:1.2: Entity type for entity Camera 4 was not initialized!
[ 5386.150918] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 5386.152283] uvcvideo: Unable to create debugfs 2-6 directory.
[ 5386.152445] uvcvideo 2-1.1:1.4: Entity type for entity Processing 7 was not initialized!
[ 5386.160662] uvcvideo 2-1.1:1.4: Entity type for entity Camera 6 was not initialized!
[ 5386.169704] input: Intel RealSense 3D Camera R200 as /devices/70090000.xusb/usb2/2-1/2-1.1/2-1.1:1.4/input/input11
[ 5386.169901] usbcore: registered new interface driver uvcvideo
[ 5386.169904] USB Video Class driver (1.1.1)
[ 5388.070613] usb 1-2.2: usb_suspend_both: status 0
[ 5389.074594] usb 2-1.1: usb_suspend_both: status 0
[ 5468.756617] usbcore: deregistering interface driver uvcvideo
[ 5470.892831] usb 2-1.1: usb_suspend_both: status 0
[ 5471.020827] usb 1-2.2: usb_suspend_both: status 0`

Any help would be much appreciated.