realsenseai/librealsense #5378: How to create a new custom RsFrameProvider class in Unity

mintos002, 2019-12-04:
Hello!

I'm trying to make a custom class that takes the color and depth frames from the `RsProcessingPipe.cs`, and do some modifications to the frame before showing theme wiht the `RsStreamTextureRender.cs`.

Can anyone share a simple example for Unity of the script I'm trying to do?

The thing is I don't know how to extract the frames once have been processed by the `RsProcessingPipe` prefab. I would like to make a script to modify those frames and share them again so the `RsStreamTextureRender.cs` can pick them up to display them.

Thanks!

wodndb, 2019-12-05:
If you want to get raw data (byte type array) of color and detph frame, I think It is better to modify RsStreamTextureRenderer.cs and RsPointCloudRenderer.cs than create custom RsFrameProvider. You know that these classes poll frames from RealSense. So, You can easily get color and depth frame raw data modifying these classes.
You can get z value directly from pointcloud frame. However If you get z value from depth frame, You should convert byte to single. So I recommend to use point cloud if you want z value.

### summary
1. Create singleton class to save color and depth frame.
2. Copy color frame and depth frame from RsStreamTextureRenderer and RsPointCloudRenderer to the singleton class.

RealSenseCustomerSupport, 2020-01-10:
----------------------------------------------

Will you be needing further assistance with this?

closed 2020-01-20 by RealSenseCustomerSupport