realsenseai/librealsense #5379: Aligning point clouds of an object captured from one angle to other

ShettyHarapanahalli, 2019-12-04:
| | |
|---|---|
|**librealsense**|2.30.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|844212071706|
|**Firmware Version**|05.11.15.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|904523050880|
|**Firmware Update Id**|904523050880|

Aligning point clouds of an object captured from one angle to other: 

I want to capture the point cloud data of an object and transform it into another angle like I want to capture an object from low angle convert to bottom angle. [http://1.bp.blogspot.com/_eANtyAC5A08/TUfrFylazQI/AAAAAAAAABw/2IdyuZsz44Q/s320/camera-angles_clip_image002.jpg](url)  

Assume that the RS is on a robotic arm, so we need to get the bottom view always for this experiment, can we use the IMU data and calculate the bottom angle irrespective of the arm orientation.

ShettyHarapanahalli, 2019-12-04:
Setup is something like the one in the one in the link below, assume getting the top view of objects will be helpful or easy to train the network without moving the arm into different positions. 
 [https://news.developer.nvidia.com/new-nvidia-research-helps-robots-improve-their-grasp/](url)

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@ShettyHarapanahalli Are you looking for the way to transform the point cloud? If so, please check https://www.tutorialspoint.com/computer_graphics/3d_transformation.htm to see if helps. Thanks!

ShettyHarapanahalli, 2020-02-06:
Thanks a ton for the response, I have been referring to open3D [https://github.com/intel-isl/Open3D].

closed 2020-02-06 by ShettyHarapanahalli

ShettyHarapanahalli, 2020-02-11:
We can also use the pytorch3d, the example demonstrates the same - 
https://github.com/facebookresearch/pytorch3d/blob/master/docs/tutorials/camera_position_optimization_with_differentiable_rendering.ipynb