realsenseai/librealsense #5383: D345 as simple USB Camera. Applicable with Cheese? or VCL Media Player

yonglizhong, 2019-12-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | 05.12.00.00 | 
| Operating System & Version |   {Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |           4.15.0-72-generic                   | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.30.0 }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {Robot/others }                          | 

### Issue Description
<Application of D435 Camera as simple USB Camera / HI all, Would like to ask around if anyone has tried using the RealSense Camera, especially the D400 series simply with Cheese application. It is working fine on the realsense-viewer, and am receiving depth and RGB stream. However, for my application I need to use the RGB stream of the camera as a simple USB Camera for image processing purposes. I'm aware of the availability of pyrealsense2 package but if it works on Cheese, I should have no problem implementing the camera for inference without the need to change my code. Thanks!>

yonglizhong, 2019-12-05:
[image]

yonglizhong, 2019-12-05:
[image]

yonglizhong, 2019-12-10:
managed to make it work by calling the RGB camera via cv2.VideoCapture("device number") without pyrealsense2 library.
unfortunately still not the case for cheese, but I'm alright with it. Any solutions to this would be helpful and informative though.
used v4l2-ctl --all -d /dev/video<no.> to make sure I called the right one.

RealSenseCustomerSupport, 2019-12-18:
----------------------------------------------

Glad you were able to find a solution. Will you be needing any further assistance with this ticket?

Thanks

closed 2020-01-22 by RealSenseCustomerSupport