realsenseai/librealsense #5384: Non-LTS Kernel 5.0 manully get the patched problem

caicaixia, 2019-12-05:
| Required Info |
|---------------------------------|------------------------------------------- |
| Camera Model | D435i & D415 |
| Operating System & Version | Ubuntu 18.04 |
| Kernel Version (Linux Only) | 5.0.0-37-generic |
| Platform | PC |
| Language | C++ |

### Issue Description

1:  I have tried Install from Debian Package follow [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages](url).

Then run `$ realsense-viewer`.  It can open the camera but have some warnings and errors.
[image]

2: Since my kernel is 5.0, I checked the installation, it says should install from source. So I installed from the source again following [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#Troubleshooting](url)

But I have the problem when I get the patched kernel: `./scripts/patch-realsense-ubuntu-lts.sh`
I tried both master and development branch, all have the problem.  Please check here:

[image]

I already googled the problem, it seems like the problem of the kernel.  I also tried all the solutions mentioned in the git and other websites, all is not working.
In some github issues,  the developer @ev-mp mentioned, kernel 5.0 is supported, but I still get this errors, why?

run `dmesg`, it shows:
`[ 1174.690912] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
`

Issues I checked and already tried:
[https://github.com/IntelRealSense/librealsense/issues/5229](url)
[https://github.com/IntelRealSense/librealsense/issues/5231](url)

MartyG-RealSense, 2019-12-05:
I hope that the link below will be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234

ev-mp, 2019-12-11:
@caicaixia hello, the `connection refused` error is a common indication of a firewall.  In order to run the script properly you should use proxy server (if available), use a non-monitored internet connection or use a separate PC to download the kernel sources and then transfer them to the original PC. 
The second warning regarding `xhci_patch` was also handled in the latest release, so using script from the latest v2.31.0 and having a properly configured proxy will allow to patch the kernel successfully (verified 5.0.0-37-generic)

ev-mp, 2019-12-17:
@caicaixia hello, did you manage to resolve the connection issue and apply the kernel patch ?

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

Hi @caicaixia, Do you need further help on this?

ev-mp, 2020-01-06:
No response received

closed 2020-01-06 by ev-mp