realsenseai/librealsense #5388: Google Pixel 2 Android 10 doesn't seem to give USB permission to the Realsense camera

nixone, 2019-12-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Android 10 Google Pixel 2  | 
| Kernel Version (Linux Only)    |  Android kernel version: 4.4.177-g7186396bf2cc                       | 
| Platform                                 | Android |
| SDK Version                            | 2.30.0                           | 
| Language                            |  Kotlin                         | 
| Segment			|  Smartphone                         | 

### Issue Description

TL;DR Android 10 doesn't seem to give USB permission to the Realsense camera. 

I debugged the code from https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android, and it reaches the same problem as my code. The `UsbManager` detects the device, can list information etc. `UsbManager` tells us that we do not have permission to use the device (`usbManager.hasPermission(device) == false`). The code requests the permission (`usbManager.requestPermission(device, intent)`). The intent is received, but `intent.getBooleanExtra(UsbManager.EXTRA_PERMISSION_GRANTED, false)` returns `false`, therefore the permission was not granted. In the meantime, nothing happened in the UI, no permission was actually asked from the user.

My code that demonstrates this problem:
Activity that handles everything: https://github.com/nixone/RealsenseTest/blob/develop/app/src/main/java/sk/nixone/realsensetest/MainActivity.kt
Manifest: https://github.com/nixone/RealsenseTest/blob/develop/app/src/main/AndroidManifest.xml

This code above ^^ never reaches the point where `usbManager.hasPermission(device)` would be true.

I am well aware this might be an Android version issue, or Pixel specific issue, but nevertheless it's forbidding me to use the Android wrapper for librealsense with the depth camera.

tunjid, 2019-12-06:
I have the same issue on my Pixel 3. How were you able to debug with the camera plugged in?

tunjid, 2019-12-06:
I think it may be a default firmware issue. I used IntelRealSenseViewer to reinstall the firmware, and it works on my Pixel 3 now. I also have the D435i, and my Pixel 3 is running Android 10.

nixone, 2019-12-06:
> I have the same issue on my Pixel 3. How were you able to debug with the camera plugged in?

@tunjid You can connect ADB through wifi, for example, USB cable isn't necessary: https://futurestud.io/tutorials/how-to-debug-your-android-app-over-wifi-without-root

Could you tell me, what firmware does your camera have?

nixone, 2019-12-06:
@tunjid tried to hardware reset, also tried to install latest firmware, it didn't help. I still have 05.11.15.00

tunjid, 2019-12-06:
@nixone I have the same firmware: 05.11.15.00. Have you tried using the other modules in the project? The capture module works very well, the camera module for some reason can't find the device. Permissions have been granted and work fine in both.

tunjid, 2019-12-08:
@nixone if you're still having problems, maybe you can try this PR: https://github.com/IntelRealSense/librealsense/pull/5395

It uses target SDK 29, and migrates the project to Androidx. Everything worked flawlessly for me here.

nixone, 2020-01-03:
Okay, I didn't find a solution yet, but a bit of an insight. I ran the same code for testing the USB permission (not even librealsense, just USB manager handling) on the Huawei P20 lite (Android 9), and it worked right away. The same code doesn't work on my Pixel 2 with Android 10. I see I have to git a little deeper into my phone.

ghlcd, 2020-01-06:
I have the same issue on HUAWEI Mate30(Android 10)，but i can connect successfully on Android 9.
I see in the logs:
```
2020-01-06 14:59:07.750 8208-8531/com.intel.realsense.camera E/UsbManager: exception in UsbManager.openDevice
    java.lang.SecurityException: User has not given 10166/com.intel.realsense.camera permission to access device /dev/bus/usb/002/004
```
Looking forward to your answer.

ghlcd, 2020-01-07:
> I have the same issue on HUAWEI Mate30(Android 10)，but i can connect successfully on Android 9.
> I see in the logs:
>
> ```
> 2020-01-06 14:59:07.750 8208-8531/com.intel.realsense.camera E/UsbManager: [...]

[java_example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example) and [native_example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/native_example) can connect successfully on Android 10.You can try.

ghlcd, 2020-01-07:
> Okay, I didn't find a solution yet, but a bit of an insight.
> 
> I ran the same code for testing the USB permission (not even librealsense, just USB manager handling) on the Huawei P20 lite (Android 9), and it worked right away. The same code doesn't work on my Pixel 2 with Android 10. I see I have to git a little deeper into my phone.
> 
> I also tried to connect other device to my pixel to test the OTG (if the connector is not bad, or if at all OTG works on my pixel). I connected a mouse, on the first try, it worked.
> 
> When having the camera connected to Pixel and then trying to switch role in USB settings to "Controlled by this device", I see in the logs:
> 
> ```
> 2020-01-03 21:09:20.214 1295-1325/? D/UsbDeviceManager: Clear notification
> 2020-01-03 21:09:27.207 904-938/? I/android.hardware.usb@1.1-service.wahoo: uevents wait timedout
> 2020-01-03 21:09:27.211 1295-1552/? E/UsbPortManager: port0 role switch failed
> ```
> 
> Digging further.
> 
> If anyone feels interested, here is a full logcat during me trying to switch control from device to pixel (from non-otg to otg):
> 
> ```
> 2020-01-03 21:14:32.886 921-957/? I/VSC: @ 307796.188: [BINNED_BRIGHTNESS] Changing state to 2 at 45652188 ms
> 2020-01-03 21:14:34.146 17659-17659/? D/UsbManager: setPortRoles Package:com.android.settings
>     
>     --------- beginning of system
> 2020-01-03 21:14:34.146 1295-4274/? I/UsbPortManager: Setting USB port mode and role: portId=port0, currentMode=dfp, currentPowerRole=source, currentDataRole=host, newMode=ufp, newPowerRole=sink, newDataRole=device
> 2020-01-03 21:14:34.146 1295-4274/? E/UsbPortManager: Trying to set the USB port mode: portId=port0, newMode=ufp
> 2020-01-03 21:14:34.148 904-938/? I/android.hardware.usb@1.1-service.wahoo: filename write: /sys/class/typec/port0/port_type role:sink
> 2020-01-03 21:14:34.149 904-1615/? I/android.hardware.usb@1.1-service.wahoo: uevent received DEVTYPE=typec_partner
> 2020-01-03 21:14:34.149 904-1615/? I/android.hardware.usb@1.1-service.wahoo: port0
> 2020-01-03 21:14:34.149 904-1615/? I/android.hardware.usb@1.1-service.wahoo: connected:0 canChangeMode:1 canChagedata:0 canChangePower:0
> 2020-01-03 21:14:34.150 1295-1552/? I/UsbPortManager: ClientCallback V1_1: port0
> 2020-01-03 21:14:34.150 1295-1325/? I/UsbPortManager: USB port changed: port=UsbPort{id=port0, supportedModes=dualsupportedContaminantProtectionModes=0supportsEnableContaminantPresenceProtection=falsesupportsEnableContaminantPresenceDetection=false, status=UsbPortStatus{connected=false, currentMode=none, currentPowerRole=no-power, currentDataRole=no-data, supportedRoleCombinations=[no-power:no-data], contaminantDetectionStatus=0, contaminantProtectionStatus=0}, canChangeMode=true, canChangePowerRole=false, canChangeDataRole=false, connectedAtMillis=0, lastConnectDurationMillis=37487
> 2020-01-03 21:14:34.156 12668-12668/? D/ModemService: Received: android.hardware.usb.action.USB_PORT_CHANGED
> 2020-01-03 21:14:34.156 12668-12668/? D/ModemService: USB Port Status: UsbPortStatus{connected=false, currentMode=none, currentPowerRole=no-power, currentDataRole=no-data, supportedRoleCombinations=[no-power:no-data], contaminantDetectionStatus=0, contaminantProtectionStatus=0}
> 2020-01-03 21:14:34.157 1295-1716/? D/UsbHostManager: Removed device at /dev/bus/usb/001/002: Intel(R) RealSense(TM) Depth Camera 435i
> 2020-01-03 21:14:34.157 1295-1716/? I/UsbAlsaManager: USB Audio Device Removed: null
> 2020-01-03 21:14:34.158 12668-12668/? D/ModemService: ModemService started
> 2020-01-03 21:14:34.158 12668-12692/? D/ModemService: CheckUpdateRunnable started
> 2020-01-03 21:14:34.158 12668-12692/? D/ModemService: Tuner by USB: enabled
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbDescriptor: UNDERRUN t:0x24 r: 8 < l: 14
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbDescriptor: UNDERRUN t:0x24 r: 12 < l: 18
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbDescriptor: UNDERRUN t:0x24 r: 7 < l: 13
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbDescriptor: OVERRRUN t:0x25 r: 8 > l: 5
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbDescriptor: UNDERRUN t:0x24 r: 7 < l: 15
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x10
> 2020-01-03 21:14:34.159 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 28 type:0x24
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x11
> 2020-01-03 21:14:34.159 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 30 type:0x24
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x11
> 2020-01-03 21:14:34.159 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 42 type:0x24
> 2020-01-03 21:14:34.159 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x11
> 2020-01-03 21:14:34.160 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 38 type:0x24
> 2020-01-03 21:14:34.160 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x11
> 2020-01-03 21:14:34.160 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 30 type:0x24
> 2020-01-03 21:14:34.160 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0xd
> 2020-01-03 21:14:34.160 1295-1716/? I/UsbDescriptorParser: Unknown Descriptor len: 6 type:0x24
> 2020-01-03 21:14:34.160 1295-1716/? W/UsbACInterface: Unknown Audio Streaming Interface subtype:0x10
[... cut ...]
> 2020-01-03 21:14:45.427 2594-28029/? I/Authzen: [DeviceStateSyncManager] The server is in sync with current state. Nothing to do
> 2020-01-03 21:14:45.432 9660-9713/? W/SearchServiceCore: Abort, client detached.
> 2020-01-03 21:14:46.170 855-855/? I//vendor/bin/hw/android.hardware.health@2.0-service.wahoo: SRAM data: 2372080
> ```

[java_example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example) and [native_example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/native_example) can connect successfully on Android 10.You can try.

ghlcd, 2020-01-09:
> I have the same issue on HUAWEI Mate30(Android 10)，but i can connect successfully on Android 9.
> I see in the logs:
>
> ```
> 2020-01-06 14:59:07.750 8208-8531/com.intel.realsense.camera E/UsbManager: [...]

I found that deleting the  targetSdkVersion number can connect successfully. @nixone

nixone, 2020-01-16:
@ghlcd your problem seemed to return also some meaningful message. Mine fails during switching between non-OTG and OTG with camera attached. That's the problem.

New knowledge: I tested on different Pixel 2 (same Android version) to eliminate for specific device problem.

RealSenseCustomerSupport, 2020-03-24:
----------------------------------------------

@nixone Sorry for late response. Did you still have this issue? Looking forward to your update. Thanks!

nixone, 2020-03-24:
@RealSenseCustomerSupport yes this is still valid. Apparently this happens on certain groups of devices / Androids. As mentioned above, Google Pixel 2 and Google Pixel 3 had problems on Android 10, I tested with Huawei P10 Lite with Android 9, it worked fine. I unfortunately don't have more options to test.

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@nixone If the issue happens on some types of devices, I'm afraid the issue is not librealsense related and you might need check if any mobile configurations caused this.

nixone, 2020-03-27:
@RealSenseCustomerSupport I would agree with you, if it would work on Android flagship devices (Google Pixel 2, Google Pixel 3, ...) and it wouldn't work on some exotic devices which barely usually work by Android specification. You shouldn't be dismissing Google Pixel devices as "some types of devices".

Yes, the fact that it doesn't work on these devices could mean that these devices do something wrong, but it could as well mean that the firmware of the camera is doing something in unusual way and it causes it to work somewhere but not work elsewhere.

Could someone at least investigate with more devices to get the idea, if it is dependent on the Android version (10), or specifically it is a Google Pixel problem?

Murded, 2020-03-31:
> @RealSenseCustomerSupport I would agree with you, if it would work on Android flagship devices (Google Pixel 2, Google Pixel 3, ...) and it wouldn't work on some exotic devices which barely usually [...]

This is also the case on the Huawei Honor View 20 (Android 10) which worked fine when it was running Android 9.  setting targetSdkVersion to an earlier version seems to fix the issue (I did targetSdkVersion 27)

RealSenseCustomerSupport, 2020-04-16:
----------------------------------------------

@Murded Thanks for sharing!
@nixone Could you please try the solution(set targetSdkVersion to an earlier version) provided by Murded to see if it helps?  Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@nixone Any update from your side? Thanks!

nixone, 2020-05-03:
Hi @RealSenseCustomerSupport ! I just tested it today. I updated the camera firwmare using the RealSenseViewer to 05.12.03.00, it resolved the issue for me :)

Just for clarification, this was most likely a firmware issue. Other apps in Play Store (for example https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=en) didn't work for me either before. After upgrading the firmware, all works, my apps, apps in Play store as well. If you want, you can close this issue.

RealSenseCustomerSupport, 2020-05-06:
----------------------------------------------

@nixone Glad to hear that resolved. Thanks for the update!

closed 2020-05-06 by RealSenseCustomerSupport