realsenseai/librealsense #5389: librealsense2-dev broken symlink in build 2.30.0-0~realsense0.1757

ihadzic, 2019-12-05:
apt update just broke me by pulling librealsense2 and librealsense2-dev, version 2.30.0-0~realsense0.1757. I ended up with broken `/usr/lib/x86_64-linux-gnu/librealsense2.so` symlink. It points to `librealsense2.so.2.31`, but there is no 2.31. The distributed .so file is still 2.30. Downgrading to `2.30.0-0~realsense0.1693` makes the problem go away

andrejpan, 2019-12-06:
I am confirming the same issue on Ubuntu 16.04 machine.

mickeyouyou, 2019-12-07:
Same situation on platform Ubuntu 18.04. Which will throw a build message with my application like couldn’t find -lrealsense2

ev-mp, 2019-12-08:
@ihadzic , @mickeyouyou  thank you bringing it to out attention - we're looking into it.

ev-mp, 2019-12-08:
An update - the root cause was isolated (unwarranted build and sync) and rectified, the incomplete versions were removed from PPA. Now the server provides the stable versions v2.30.0 both for Ubuntu 16 and 18 (builds 1693/1694 respectively).
Note that the release of v2.31 is underway, so those Debian will be superseded with the new version in the next 24 hours.

mickeyouyou, 2019-12-08:
Good

sumitsarkar1, 2019-12-08:
I cannot link realsense2 with my C++ QtCreator IDE. I have the following line for linking the realsense2 library `LIBS += -L/usr/lib/x86_64-linux-gnu -lrealsense2 \` . However it says "could not find -lrealsense2" although I have checked manually
[image]
please help me

ev-mp, 2019-12-09:
@sumitsarkar1 hello,  please clarify which debian package are you using ?
```sh
dpkg -l | grep librealsense2-dev | tr -s [:blank:] | cut -d' '  -f 3
```
Can you also check the actual symlinks ?
```sh
ll /usr/lib/x86_64-linux-gnu/librealsense2.so
ll /usr/lib/x86_64-linux-gnu/librealsense2.so.2.30
```

sumitsarkar1, 2019-12-12:
1.) package : 2.30.0-0~realsense0.1757
2.) this is th output of symlink check
    a.) for ll /usr/lib/x86_64-linux-gnu/librealsense2.so I got : lrwxrwxrwx 1 root root 21 Dec  5 19:47 /usr/lib/x86_64-linux-gnu/librealsense2.so -> librealsense2.so.2.31
    b.) l /usr/lib/x86_64-linux-gnu/librealsense2.so.2.30  :  lrwxrwxrwx 1 root root 23 Dec  5 19:47 /usr/lib/x86_64-linux-gnu/librealsense2.so.2.30 -> librealsense2.so.2.30.0

sumitsarkar1, 2019-12-13:
there is no librealsense2.so.2.31 on my machine (Ubuntu 16.04)

ev-mp, 2019-12-17:
@sumitsarkar1 , did you manage to remove 1757 build and replace it with a proper version as indicated above?

sumitsarkar1, 2019-12-18:
no I didnt downgrad it. If downgrading is the solution then I can go ahead and do that.

ev-mp, 2019-12-18:
@sumitsarkar1, the solution is not to use build 1757. 
The PPA is updated so it is up to you to choose whether to upgrade or downgrade. 

Let us know if you need further assistance.

ev-mp, 2020-01-19:
Assume resolved.

closed 2020-01-19 by ev-mp

TheButlah, 2020-02-26:
@ev-mp I am having this same issue now, but this time with the latest apt package of realsense. Here are my symlinks below:
[image]

Note that the `librealsense2.so` symlinks to a nonexistent `librealsense2.so.2.33`