realsenseai/librealsense #5391: Depth not the same with different method

daothanlong11, 2019-12-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.11.15.00) | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0                          | 
| Language                            | C++                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to compare the result of distance with 2 method at the same point but it give me not the same, here is my code

#include <librealsense2/rs.hpp>
#include <librealsense2/rsutil.h>
#include <iostream>
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;

cv::Mat depth_frame_to_meters(const rs2::pipeline& pipe, const rs2::depth_frame& f);

int main()
{
    const auto window_name = "des";
    namedWindow(window_name,WINDOW_AUTOSIZE);
    rs2::pipeline pip;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
    pip.start(cfg);

    while(waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset frame = pip.wait_for_frames();
        auto depth_frame = frame.get_depth_frame();

        

        Mat depth = depth_frame_to_meters(pip,depth_frame);
        cout<<"1: "<<depth.at<float>(180,101)<<endl;
        cout<<"2: "<<depth_frame.get_distance(180,101)<<endl;
        
        
    }

    return EXIT_SUCCESS;
}

cv::Mat depth_frame_to_meters(const rs2::pipeline& pipe, const rs2::depth_frame& f)
{
    using namespace cv;
    using namespace rs2;

    int w = f.get_width();
    int h = f.get_height();
    Mat image(Size(630,480),CV_16UC1,(void*)f.get_data(),Mat::AUTO_STEP);

    Mat dm = image;
    dm.convertTo(dm, CV_64F);
    auto depth_scale = pipe.get_active_profile()
        .get_device()
        .first<depth_sensor>()
        .get_depth_scale();
    cout<<"depth scale: "<<depth_scale<<endl;
    dm = dm * depth_scale;
    return dm;
}

So which one is correct. Hope to see your answer:))

ev-mp, 2019-12-09:
There is a typo in:
>  int w = f.get_width();
    int h = f.get_height();
Mat image(Size( **640** ~~630~~ ,480),CV_16UC1,(void*)f.get_data(),Mat::AUTO_STEP);

It seem thought as it were meant to use `w` and `h` instead.
Other than that  it should produce similar results

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Does that resolve your issue daothanlong11?

Thanks

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

If we don’t hear from you in 7 days, this issue will be closed.

daothanlong11, 2020-02-23:
> If we don’t hear from you in 7 days, this issue will be closed.

sorry for late reply. Yes, my issue is solve, the problem is in first method, need to switch width and height, because of using .at()

closed 2020-02-23 by daothanlong11