realsenseai/librealsense #5394: T265: New (old?) USB problems on macOS Catalina

neilyoung, 2019-12-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.879 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0                          | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

Hmm, I already had made my peace with these issues, but today I downloaded a fresh copy of the entire repo, switched to development branch, applied PR #5213, compiled and installed it all again.

It is sometimes hard to launch my T265 from cold boot. Nothing helps: No plug/unplug, no restart. Twice a start of realsense-viewer helped. From that on the things did roll.

Before I got things like this (I have enabled severity console logging in pose.cpp):

```
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
 06/12 22:55:42,569 INFO [0x111dffdc0] (tm-info.cpp:51) Picked 0/3 devices
 06/12 22:55:42,572 INFO [0x111dffdc0] (tm-info.cpp:51) Picked 0/3 devices
 06/12 22:55:42,572 WARNING [0x111dffdc0] (rs.cpp:1618) null pointer passed for argument "profile"
RealSense error calling rs2_config_resolve(config:0x7f8289cbc180, pipe:0x7f8289ca3780):
    No device connected

```

After re-plug then continuoulsy:

```
/Documents/librealsense/build/examples/pose $ ./rs-pose 
 06/12 22:53:16,680 INFO [0x106d57dc0] (tm-info.cpp:51) Picked 1/4 devices
 06/12 22:53:16,684 INFO [0x106d57dc0] (tm-device.cpp:1898) Serial: 0000929122110112
 06/12 22:53:16,684 INFO [0x106d57dc0] (tm-device.cpp:1900) Connection type: 3.1
 06/12 22:53:16,684 INFO [0x106d57dc0] (tm-device.cpp:1906) Firmware version: 0.2.0.879
 06/12 22:53:16,685 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 207714271: [0x/1:386] TM2 version 0.2.0.879, sku 2155
 06/12 22:53:16,685 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 375081905: [0x/29:190] LeonIPCRxReassignSinkThread status: 0
 06/12 22:53:16,685 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 375188185: [0x/29:192] LOS started
 06/12 22:53:16,686 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 375257871: [0x/29:193] LeonIPC message size is 1
 06/12 22:53:16,686 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 91700253: [0x/29:190] LeonIPCRxReassignSinkThread status: 0
 06/12 22:53:16,686 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 91844193: [0x/29:192] LRT started
 06/12 22:53:16,686 WARNING [0x70000d04e000] (tm-device.cpp:1378) T265 FW message: 91916898: [0x/29:193] LeonIPC message size is 1
 06/12 22:53:16,688 INFO [0x106d57dc0] (tm-device.cpp:474) Added a fisheye sensor id: 0
 06/12 22:53:16,689 INFO [0x106d57dc0] (tm-device.cpp:474) Added a fisheye sensor id: 1
 06/12 22:53:16,689 INFO [0x106d57dc0] (tm-device.cpp:506) Added a gyro sensor id: 0 at 200Hz
 06/12 22:53:16,689 INFO [0x106d57dc0] (tm-device.cpp:506) Added a accel sensor id: 0 at 62Hz
 06/12 22:53:16,689 INFO [0x106d57dc0] (tm-device.cpp:509) Skipped a velocimeter stream
 06/12 22:53:16,689 INFO [0x106d57dc0] (tm-device.cpp:509) Skipped a velocimeter stream
 06/12 22:53:16,697 INFO [0x106d57dc0] (tm-device.cpp:599) Request for stream type 8 with stream index 0
 06/12 22:53:16,697 INFO [0x106d57dc0] (tm-device.cpp:644) Activated 1/5 streams
 06/12 22:53:18,212 ERROR [0x106d57dc0] (types.h:304) open(...) unknown error starting TIMEOUT
 06/12 22:53:18,212 WARNING [0x106d57dc0] (rs.cpp:1618) null pointer passed for argument "profile"
 06/12 22:53:28,375 WARNING [0x70000d04e000] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x82, error: Operation timed out, err. num: 60
 06/12 22:53:28,375 ERROR [0x70000d04e000] (tm-device.cpp:1983) Bulk response error RS2_USB_STATUS_TIMEOUT
 06/12 22:53:28,375 INFO [0x70000d04e000] (tm-device.cpp:1407) Got bad response, stopping log_poll
 06/12 22:53:38,535 WARNING [0x70000d0d1000] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: Operation timed out, err. num: 60
 06/12 22:53:38,535 ERROR [0x70000d0d1000] (tm-device.cpp:1967) Bulk request error RS2_USB_STATUS_TIMEOUT
 06/12 22:53:38,535 INFO [0x70000d0d1000] (tm-device.cpp:1422) Got bad response, stopping time sync
 06/12 22:53:48,549 WARNING [0x106d57dc0] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: Operation timed out, err. num: 60
 06/12 22:53:48,549 ERROR [0x106d57dc0] (tm-device.cpp:1967) Bulk request error RS2_USB_STATUS_TIMEOUT
RealSense error calling rs2_pipeline_start_with_config(pipe:0x7fdb8f4085e0, config:0x7fdb8f40b470):
    open(...) unknown error starting TIMEOUT

~/Documents/librealsense/build/examples/pose $ 

```

Only realsense-viewer came to help. After starting the device there, all C and python attempts work. Until the device was powered off by unplugging. Then the same sequence restarted.

I'm sorry, I had a pretty "old" development branch until today and I thought it would have been a good idea to start from scratch. Until today a lot of my USB problems were gone with PR #5213 . I don't currently know, what happened.

neilyoung, 2019-12-06:
...plus building the master branch dies at 14% with linker error:

```
[ 14%] Linking CXX shared library pybackend2.cpython-37m-darwin.so
[ 14%] Built target pybackend2
make: *** [all] Error 2

```

neilyoung, 2019-12-06:
OK, I'm not sure, if this makes sense: Cloning repo, checkout development, apply #5213, compile, install ---> fail

Cloning repo, checkout development, compile, install, apply #5213, compile, install  ---> works.

Just one (also known issue): Starting a new run too early leads to an error:

```
Device Positio^C 0.000 0.001 -0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Device Position: -0.002 0.002 -0.000 (me^Crs)
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
De^Cce Position: -0.001 -0.000 -0.001 (meters)
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Device Position:^C0.000 0.000 0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
RealSense error calling rs2_pipeline_start_with_config(pipe:0x7ffbdb508f70, config:0x7ffbdb508f80):
    open(...) failed to configure streams. T265 is running!
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Device Position: -0^C00 0.000 -0.001 (meters)
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Device Position: -0.001 -0.000 -0.000 (met^Cs)
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
RealSense error calling rs2_pipeline_start_with_config(pipe:0x7fa42c40ef60, config:0x7fa42c4076a0):
    open(...) failed to configure streams. T265 is running!

Well, I don't know. Either I have now a lucky phase, or the way, how the PR has to be applied, has to be as described. 

```

neilyoung, 2019-12-06:
Disregard

neilyoung, 2019-12-08:
Here a small collection of consecutive runs of `rs-pose`, stopped by CTRL-C and immediately started. My observation with latest repo, USB and macOS Catalina is:

1) Master: Does not work at all reliably
2) Development: Does not work if not patched with #5213 
3) If patched with #5213: Does work (at least better than Master or w/o patch), but far from working stable and reliably. See below.

```
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device Posit^Cn: 0.000 0.000 -0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device ^Csition: -0.000 0.000 0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device Position: 0.^C0 0.000 -0.001 (meters)))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
RealSense error calling rs2_config_resolve(config:0x7facd8d00760, pipe:0x7facd8d00720):
    No device connected
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
D^Cice Position: -0.001 -0.000 0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device Position: -0.000 0.000 0.000^Cmeters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
RealSense error calling rs2_config_resolve(config:0x7fdfce901cb0, pipe:0x7fdfce901530):
    No device connected
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device Position: -0.000 0.001 ^C000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Device Pos^Cion: -0.001 0.001 0.001 (meters)))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
De^Cce Position: -0.000 0.004 0.000 (meters))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
libc++abi.dylib: terminating
Abort trap: 6
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Dev^Ce Position: 0.000 0.000 -0.001 (meters)))
~/Documents/librealsense/build/examples/pose $ ./rs-pose
libc++abi.dylib: terminating
Abort trap: 6
~/Documents/librealsense/build/examples/pose $ 

```

neilyoung, 2019-12-08:
Even if it works, it is decorated with a couple of disturbing warnings :)

```
08/12 09:22:59,859 INFO [0x11d35edc0] (tm-info.cpp:51) Picked 1/4 devices
 08/12 09:22:59,864 INFO [0x11d35edc0] (tm-device.cpp:1898) Serial: 0000929122110112
 08/12 09:22:59,864 INFO [0x11d35edc0] (tm-device.cpp:1900) Connection type: 3.1
 08/12 09:22:59,864 INFO [0x11d35edc0] (tm-device.cpp:1906) Firmware version: 0.2.0.879
 08/12 09:22:59,867 INFO [0x11d35edc0] (tm-device.cpp:474) Added a fisheye sensor id: 0
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 7125737913: [0x/5:100] mempool out of memory: video
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 7158995663: [0x/5:100] mempool out of memory: video
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 7159210150: [0x/5:100] mempool out of memory: video
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 7905721400: [0x/28:616] ASSERTION ret != RTEMS_TIMEOUT failed: sema
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 7905721400: [0x/28:616] phore of write endpoint 1 was not released
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 6517478620: [0x/31:546] Can't get stage test
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 6527461496: [0x/31:546] Can't get stage test
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 6537433891: [0x/31:546] Can't get stage test
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 6547570915: [0x/31:546] Can't get stage test
 08/12 09:22:59,868 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 6557459168: [0x/31:546] Can't get stage test
 08/12 09:22:59,868 INFO [0x11d35edc0] (tm-device.cpp:474) Added a fisheye sensor id: 1
 08/12 09:22:59,868 INFO [0x11d35edc0] (tm-device.cpp:506) Added a gyro sensor id: 0 at 200Hz
 08/12 09:22:59,868 INFO [0x11d35edc0] (tm-device.cpp:506) Added a accel sensor id: 0 at 62Hz
 08/12 09:22:59,869 INFO [0x11d35edc0] (tm-device.cpp:509) Skipped a velocimeter stream
 08/12 09:22:59,869 INFO [0x11d35edc0] (tm-device.cpp:509) Skipped a velocimeter stream
 08/12 09:22:59,871 INFO [0x11d35edc0] (tm-device.cpp:599) Request for stream type 8 with stream index 0
 08/12 09:22:59,871 INFO [0x11d35edc0] (tm-device.cpp:644) Activated 1/5 streams
s
```

At runtime every second this warning appears:

```
08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1213342921: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1223383580: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1233317121: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1243334811: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1253428341: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1263314201: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1273581853: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1283330970: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1293317970: [0x/31:546] Can't get stage test
 08/12 09:23:00,713 WARNING [0x700008f9f000] (tm-device.cpp:1378) T265 FW message: 1303322130: [0x/31:546] Can't get stage test

```

bfulkers-i, 2019-12-11:
@neilyoung I pushed a few changes to #5213 that apply to this (e.g. 724c7e1) and for convenience I added #5426 to that branch also (which I believe will solve #5401). Let us know if that solves your issue, and feel free to close any you think are solved.

neilyoung, 2019-12-11:
@bfulkers-i Thanks. I have cloned a fresh copy of the repo, switched to development and applied #5213. Then  #5426 .

While doing the latter I got an error:

```
~/Documents/librealsense $ git pull origin pull/5426/head
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 6 (delta 4), reused 5 (delta 4), pack-reused 0
Unpacking objects: 100% (6/6), done.
From https://github.com/IntelRealSense/librealsense
 * branch                refs/pull/5426/head -> FETCH_HEAD
CONFLICT (modify/delete): third-party/libtm/fw/CMakeLists.txt deleted in HEAD and modified in 610aa744e2fcdde8f68d44ab8c4a8b104482e52b. Version 610aa744e2fcdde8f68d44ab8c4a8b104482e52b of third-party/libtm/fw/CMakeLists.txt left in tree.
Automatic merge failed; fix conflicts and then commit the result.

```

Would you mind to share, how to recover from this?

TIA

bfulkers-i, 2019-12-11:
No need to apply #5426, I already did that in 56cf379, you should simply checkout #5213 and build that. I periodically also rebase and force-push #5213 on the `development` branch so that it will be easy to merge.

neilyoung, 2019-12-11:
@bfulkers-i OK, that looks very promising. First 10-15 runs no problems. I will continue to test.
Thanks

neilyoung, 2019-12-11:
@bfulkers-i Just one thing, which does still not work: You can't successfully save a map w/o making a pause between `pipe.stop()` and `export_localization_map()` I'm waiting 5 s, this seems to be ok.

bfulkers-i, 2019-12-11:
@neilyoung thanks for the reminder, pull and try again, I added 467c621 which I believe fixes that (no wait required).

neilyoung, 2019-12-11:
@bfulkers-i Thanks. Will do later the day and report

neilyoung, 2019-12-11:
@bfulkers-i Doesn't except anymore, but doesn't save either :)

```
                std::vector<uint8_t> map = pose_sensor.export_localization_map();
                printf("map exported, result: %lu bytes\n", map.size());

```

Result:

`map exported, result: 0 bytes
`

Result with wait:

```
stopping pipe, saving map
pipe stopped ... sleeping 5s
map exported, result: 67339 bytes

```

neilyoung, 2019-12-12:
@bfulkers-i I thought the 0 bytes would be explainable because I didn't move much with my first tests, so I omitted the timeout again and walked a little bit longer (20s). Same result.

**W/o making a pause after stop the map is always empty.**

bfulkers-i, 2019-12-12:
@neilyoung so far I'm unable to reproduce this behavior, can you possibly provide more code to show exactly what you are doing? My test is:
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#include <librealsense2/rs.hpp>
#include <iostream>
#include <fstream>
#include <thread>
#include <chrono>

int main(int argc, char * argv[])
{
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pose_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();

    pipe.start(cfg);
    std::cerr << "Move the device...\n";
    std::this_thread::sleep_for(std::chrono::seconds(10));
    std::cerr << "Stop\n";
    pipe.stop();
    auto map = pose_sensor.export_localization_map();
    std::cerr << "Map is " << map.size()  << " bytes\n";
    return EXIT_SUCCESS;
}
```

neilyoung, 2019-12-12:
@bfulkers-i Let me run your code... One second

neilyoung, 2019-12-12:
I admit your code works fine. I have altered it and added some wireframes of my code and I can stop the entire engine. Mostly if I play a bit with set_static_node. You need to run this at least 3-5 times until the problem comes.

Hit 'n' for setting a static node where you are, hit 's' to stop and save the map.

```
#include <stdlib.h>
#include <sys/select.h>
#include <termios.h>
#include <unistd.h>
int kbhit() {
    struct timeval tv = {0L, 0L};
    fd_set fds;
    FD_ZERO(&fds);
    FD_SET(0, &fds);
    return select(1, &fds, NULL, NULL, &tv);
}

int getch() {
    int r;
    unsigned char c;
    if ((r = read(0, &c, sizeof(c))) < 0) {
        return r;
    } else {
        return c;
    }
}

int main(int argc, char* argv[]) {
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pose_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();
    pose_sensor.set_option(RS2_OPTION_ENABLE_MAP_PRESERVATION, 1);

    pipe.start(cfg);
    std::cerr << "Move the device...\n";

    bool got_static_node = false;
    while (true) {
        auto frames = pipe.wait_for_frames();
        auto frame = frames.first_or_default(RS2_STREAM_POSE);
        auto pose_data = frame.as<rs2::pose_frame>().get_pose_data();

        rs2_vector translation;
        rs2_quaternion rotation;
        if (!got_static_node && pose_sensor.get_static_node("test", translation, rotation)) {
            printf("get_static_node ok\n");
            got_static_node = true;
        }

        if (kbhit()) {
            int c = getch();

            if (c == 'n') {         // Set static node
                if (pose_data.tracker_confidence == 3) {
                    int result = pose_sensor.set_static_node("test", pose_data.translation, pose_data.rotation);
                    printf("set_static_node result %d\n", result);
                    if (result == true) {
                        got_static_node = false;
                    }
                }
                else {
                    printf("insufficient confidence %d\n", pose_data.mapper_confidence);
                }
            }
            if (c == 's') {         // Stop
                break;
            }
        }

    }
    //std::this_thread::sleep_for(std::chrono::seconds(10));
    std::cerr << "Stop\n";
    pipe.stop();
    auto map = pose_sensor.export_localization_map();
    std::cerr << "Map is " << map.size() << " bytes\n";
    return EXIT_SUCCESS;
}
```

neilyoung, 2019-12-12:
@bfulkers-i I never got that "0" bytes again. Really strange, but after a couple of attempts this mostly happens:

```
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
libc++abi.dylib: terminating with uncaught exception of type rs2::error: Frame didn't arrive within 15000
Abort trap: 6

```

neilyoung, 2019-12-12:
@bfulkers-i Yepp, now also with "your" code. Note how many attempts it took.

And I added some other options, don't know if it makes a diff, so the whole script looks like so now:

```

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#include <chrono>
#include <fstream>
#include <iostream>
#include <librealsense2/rs.hpp>
#include <thread>

#include <stdlib.h>
#include <sys/select.h>
#include <termios.h>
#include <unistd.h>

int kbhit() {
    struct timeval tv = {0L, 0L};
    fd_set fds;
    FD_ZERO(&fds);
    FD_SET(0, &fds);
    return select(1, &fds, NULL, NULL, &tv);
}

int getch() {
    int r;
    unsigned char c;
    if ((r = read(0, &c, sizeof(c))) < 0) {
        return r;
    } else {
        return c;
    }
}

int main(int argc, char* argv[]) {
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pose_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();
    pose_sensor.set_option(RS2_OPTION_ENABLE_MAP_PRESERVATION, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_MAPPING, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_RELOCALIZATION, 1);

    pipe.start(cfg);
    std::cerr << "Move the device...\n";

    bool got_static_node = false;
    while (true) {
        auto frames = pipe.wait_for_frames();
        auto frame = frames.first_or_default(RS2_STREAM_POSE);
        auto pose_data = frame.as<rs2::pose_frame>().get_pose_data();

        rs2_vector translation;
        rs2_quaternion rotation;
        if (!got_static_node && pose_sensor.get_static_node("test", translation, rotation)) {
            printf("get_static_node ok\n");
            got_static_node = true;
        }

        if (kbhit()) {
            int c = getch();

            if (c == 'n') {         // Set static node
                if (pose_data.tracker_confidence == 3) {
                    int result = pose_sensor.set_static_node("test", pose_data.translation, pose_data.rotation);
                    printf("set_static_node result %d\n", result);
                    if (result == true) {
                        got_static_node = false;
                    }
                }
                else {
                    printf("insufficient confidence %d\n", pose_data.mapper_confidence);
                }
            }
            if (c == 's') {         // Stop
                break;
            }
        }

    }
    //std::this_thread::sleep_for(std::chrono::seconds(10));
    std::cerr << "Stop\n";
    pipe.stop();
    auto map = pose_sensor.export_localization_map();
    std::cerr << "Map is " << map.size() << " bytes\n";
    return EXIT_SUCCESS;
}

```

Steps to the error:

```

~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
n
set_static_node result 1
get_static_node ok
s
Stop
Map is 631355 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
s
Stop
Map is 823145 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
s
Stop
Map is 830209 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
s
Stop
Map is 652053 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
s
Stop
Map is 679467 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose
Move the device...
get_static_node ok
s
Stop
Map is 0 bytes
^C
~/Documents/librealsense/build/examples/pose $ 
```

bfulkers-i, 2019-12-12:
Thanks for the code @neilyoung, I see weird behavior in your example too. We will take a closer look and reply here.

neilyoung, 2019-12-12:
@bfulkers-i Many thanks. Good luck. Let me know, if you need my assistance

bfulkers-i, 2019-12-16:
@neilyoung We are in the process of fixing this map saving bug described above, but it looks like it will require a new version of the firmware. We will aim to update that toward the end of this week. I'm changing the title of this to reflect the new scope of this issue.

neilyoung, 2019-12-16:
@bfulkers-i Thanks for letting me know. Take your time.

neilyoung, 2019-12-18:
@bfulkers-i Two attempts with the latest #5213, 64c753b67c5dbc1299dcc4adf037c0213231b7bb.

Script from here https://github.com/IntelRealSense/librealsense/issues/5394#issuecomment-564984281

Result:

```
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Move the device...
n
set_static_node result 1
get_static_node ok
s
Stop
Map is 60405 bytes
~/Documents/librealsense/build/examples/pose $ 
~/Documents/librealsense/build/examples/pose $ ./rs-pose 
Move the device...
get_static_node ok
s
Stop
Map is 0 bytes

```

Was the 0 byte issue supposed to be fixed with this commit?

neilyoung, 2019-12-18:
Fixed

closed 2019-12-18 by neilyoung

radfordi, 2019-12-18:
I suspect you just got lucky, @neilyoung.  We'll have some fixes soon that should be much more reliable.

neilyoung, 2019-12-18:
OK, cool. I admit, after my euphoric post there have been the same issues again. But not that often as before. Keen to see the final result.