realsenseai/librealsense #5396: Could not initialize offscreen context!

himanshugupta1009, 2019-12-07:
I know this has been asked before but none of the solutions helped me. I still face this error. I am trying to use Real Sense on Ubuntu 16.04 and when I try to tun "realsense-viewer", I get this output.

I have already checked that glfw3, mesa and gtk packages are installed. 

Kindly let me know the fix for this. Thanks in advace!

dorodnic, 2019-12-10:
This error indicates general problem with your graphics drivers installation. (you will most likely get similar error from any other application using OpenGL 3+) There is no one-size-fits-all, since it depends on platform and type of GPU. For NVidia it would be something like this - [ubuntu-linux-install-nvidia-driver-latest-proprietary-driver](https://www.cyberciti.biz/faq/ubuntu-linux-install-nvidia-driver-latest-proprietary-driver/)

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Were you able to try dorodnic's recommendation?

closed 2020-01-13 by RealSenseCustomerSupport