realsenseai/librealsense #5401: T265: A safe way to hang the camera on macOS, development branch, #5213 applied

neilyoung, 2019-12-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.879 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development branch, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

Probably related to the problems reported here https://github.com/IntelRealSense/librealsense/issues/5394

Receipt to hang a T265 with the a.m. firmware and SDK under macOS Catalina:

- Take pose example
- Alter these lines

```

    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

```

like so:

```
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pose_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();
    // Options
    pose_sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_MAPPING, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_RELOCALIZATION, 1);
    pose_sensor.set_option(RS2_OPTION_ENABLE_MAP_PRESERVATION, 1);

```

It should not work anymore.

neilyoung, 2019-12-10:
No comments on this? For me this is reproducible, but my sequence may be wrong at all...

neilyoung, 2019-12-10:
And would somebody please comment on the internal defaults of these cfg settings?

```
RS2_OPTION_ENABLE_POSE_JUMPING
RS2_OPTION_ENABLE_MAPPING
RS2_OPTION_ENABLE_RELOCALIZATION
RS2_OPTION_ENABLE_MAP_PRESERVATION

```

My guess is: true, true, true, false. Is that correct?

neilyoung, 2019-12-10:
The astonishing fact: The python binding is absolutely not affected by this bug (?) I can set_options and the script runs w/o issues. It is just the CPP thing...

bfulkers-i, 2019-12-10:
Thanks for the report @neilyoung, I am able to reproduce this, we will look into it and provide a fix.

neilyoung, 2019-12-10:
@bfulkers-i Great. Thanks for the follow up.

neilyoung, 2019-12-11:
Fixed with https://github.com/IntelRealSense/librealsense/issues/5394

closed 2019-12-11 by neilyoung