realsenseai/librealsense #5403: The right way to deproject pixel to point with align to depth frame.

BadMachine, 2019-12-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (2.29.0 ?) | 
| Operating System & Version |   {Windows 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { latest }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {VR/AR }                          | 

### Issue Description
I'm trying to get accurate points in 3d space using "rs2_deproject_pixel_to_point" function.
I have 2 problems with it:
1) Sometimes, when i have to get distance to {x, y} coordinate of pixel i`m getting distance = 0, as you can realize then result of "rs2_deproject_pixel_to_point" is {x=0, y=0, z=0}
2) When im getting non-zero values of depth (in meters) the values can be in range (true, close to be true and absolutly inaccurate), i mean the distance value is jumping over 1-2 meters.

## Steps of the programm
# Grabbing frames and intrinsics
```C++

       rs2::pipeline p;
       rs2::config cfg;
		cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);
		cfg.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_RGB8);
       p.start(cfg);
while (!QThread::currentThread()->isInterruptionRequested())
		{	
			rs2::frameset frames = p.wait_for_frames();
			frames = align_to_depth.process(frames);
			rs2::depth_frame depth = frames.get_depth_frame();
			auto inrist = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();
			auto color_mat = frame_to_mat(color_frame);
			emit sendFrame(frames, inrist);

		}

```
# Getting necessary pixels in frame

[image]

# Finding 3d coordinates of those pixels
```C++

			float planarPoint3d[3];
			float x =(float)toSend.captured[i.key()].x;
                        float y = (float)toSend.captured[i.key()].y };

float pixel_distance_in_meters = depth.get_distance(x, y);
rs2_deproject_pixel_to_point(planarPoint3d, &inrist, {x,y}, pixel_distance_in_meters);
```

# RESULT

[image]

### Final questions

Am i using those function right?
Did i make mistake when align color frame to depth?
Did i get wrong intrinsics?
What am i doing wrong?)

BadMachine, 2019-12-08:
UPD: 
i applied few filters, like decimation, spatial and temporal, and got better, but still sometimes it`s "jumping" and shows wrong depth values

ev-mp, 2019-12-09:
@BadMachine , getting the final result oscillating in range of +-1m is unacceptable, so it is more likely that the data path is not configured correctly.
There is always a possibility that the device is not functioning properly, so running the `rs-depth-quality` tool and using the [accompanying white paper](https://www.intel.com/content/www/us/en/support/articles/000027019/emerging-technologies/intel-realsense-technology.html) allows to assess whether the depth characteristics are in line with the expected results

As for the reported behavior -
Having zero depth value is a legitimate input, it just mean that that the sensor was not able to find a high-confidence match for the specific [i,j] coordinate.The mentioned tool `rs-depth-quality` has a similar metric called ["Fill Rate"](https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality#fill-rate) that shows the percentage of valid (non-zero) pixels in frame, and the number is generally less than 100%.
There are methods to improve the depth quality that  affect the fill rate , such as adjusting preset, turning on and modifying the laser power. Reading the relevant [white paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) can be a good reference point.
One of the ways to reduce the number of zero points is to apply Depth decimation filter which in addition to reducing the depth resolution also delivers some "hole filling" capability.

But in any event these steps are secondary to the actual verification of the user c++ code.

I'll make certain assumptions as the snippet is neither covering the configuration part nor the RGB features/image handling. So the things to verify are:
- The alignment is from color to depth as the member `align_to_depth` suggests, i.e. you generate a synthetic RGB frame captured as if it were positioned in the Depth sensor's location. If that is correct then the intrinsic values you extract are correct as well.
- The RGB features that you extract for the skeleton tracking area calculated based on the "Aligned RGB" and not the original RGB frame.

ev-mp, 2019-12-09:
@BadMachine , looking deeper at the posted results of deprojected data I can see the depth fluctuates in range of [1.368-1.413] meter ,i.e. it seem to be +-2.5cm which approximates for 1.8% of the range. 
Note that that the D435i [specification states RMS<2%](https://www.intel.com/content/www/us/en/support/articles/000027019/emerging-technologies/intel-realsense-technology.html), so the presented values are within the boundaries.

Can you provide an example with valid depth values deviating within 1 meter range?

BadMachine, 2019-12-09:
@ev-mp Thank you for your reply!
First of all:
I applied "custom preset":
```json
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "5040",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "300",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "8",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "8",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "1",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "751",
    "param-lambdacensus": "6",
    "param-leftrightthreshold": "10",
    "param-maxscorethreshb": "2893",
    "param-medianthreshold": "796",
    "param-minscorethresha": "4",
    "param-neighborthresh": "108",
    "param-raumine": "6",
    "param-rauminn": "3",
    "param-rauminnssum": "7",
    "param-raumins": "2",
    "param-rauminw": "2",
    "param-rauminwesum": "12",
    "param-regioncolorthresholdb": "0.784736",
[... 320 lines cut ...]
coords:  QJsonObject({"x":-0.5700000000000001,"y":-0.05,"z":1.48})
x:  206 y:  229 

coords:  QJsonObject({"x":-0.51,"y":-0.05,"z":1.45})
x:  217 y:  229 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  237 y:  240 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  258 y:  250 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  289 y:  250 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  320 y:  271 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  351 y:  281 

coords:  QJsonObject({"x":0,"y":0,"z":0})
x:  392 y:  292 

coords:  QJsonObject({"x":-0.22,"y":0.32,"z":4.47})

ev-mp, 2019-12-09:
@BadMachine, 
The code that you've just posted in not the same as in the original report  - in the original snippet the color is aligned to depth
>rs2::frameset frames = p.wait_for_frames();
frames = align_to_depth.process(frames);

while in the second post the direction is reverted with depth is aligned to color:
>rs2::frameset frames = p.wait_for_frames();
frames = align_to_color.process(frames);

Thus the application flow is still not clear.
In both cases though, the critical element is still missing - the handling of RGB data.
Without seeing in code which frame is used to find tracking features it is not possible to assess whether the flow is implemented correctly.
It is obvious why aligning Depth to RGB is preferable in terms of extracting the skeleton features as the RGB image remains unchanged. Note, however that this represents a challenge as not for each RGB pixel you'll have a valid depth value, enforcing to look up for neighboring depth values.
Aligning color to depth is probably more beneficial, since even though part of the RGB data is not present in the aligned image, the pixels that do pass the filter will have 100% correspondence in the Depth data.

Please try to present the case and the issue you have in a concise and unambiguous manner, as it really not effective to try and guess the missing parts.

BadMachine, 2019-12-09:
@ev-mp 
Now i return to using align color to depth.
I'm really trying to make it clear.
So i want to figure out why in this frame i see "Right palm joint"
[image]

But depth of the joint pixel-point is 0?

Here is code that i wrote to catch the frame when depth of "right palm" is 0:

```c++

float planarPoint3d[3];
			float pix[2] = { toSend.captured[i.key()].x, toSend.captured[i.key()].y };

			float pixel_distance_in_meters = depth.get_distance(toSend.captured[i.key()].x, toSend.captured[i.key()].y);
		
			rs2_deproject_pixel_to_point(planarPoint3d, &inrist, pix, pixel_distance_in_meters);

			QJsonObject point3D;
			point3D.insert("x", QJsonValue::fromVariant(0.01 * floor(100 * planarPoint3d[0])));
			point3D.insert("y", QJsonValue::fromVariant(0.01 * floor(100 * planarPoint3d[1])));
			point3D.insert("z", QJsonValue::fromVariant(0.01 * floor(100 * planarPoint3d[2])));
			if (i.key() == "Right palm") {
				if (planarPoint3d[2] == 0)
				{
					imshow("image", toSend.input.getMat(ACCESS_READ));
					waitKey(30);
					//cv::imwrite("wrong depth.jpg", toSend.input);
				}
```

if I understand you right i should use some specific functions to get nearest pixels depth with non-zero value, right?

Also i`ll try to show you processes of my program, picture related:
[image]

I`m sorry if its still unclear.
I`m trying to explain my problem so hard.

ev-mp, 2019-12-09:
@BadMachine , thanks for the explanation.
The tracking seem quite consistent - very impressive!

For clarification :  
- The alignment takes place before sending the frame to pose estimator.
- The pose estimator extract the tracking features from the RGB image only.
- The 3d coordinates are extracted from the original depth image.
Are those assumptions correct ?

There is still something seem not right with the "Aligned Color" - I would expect that image to be much "noisier" and to have much more "zero" pixels as it is in the relevant depth image.
Can you provide the three images for verification : Original RGB, "RGB aligned to Depth" and the Depth snapshots.

When using the "Aligned RGB image make sure that the pose estimator skips the areas with no valid RGB data with zero/black pixels. And that opposing to the depth pixels where zero represents no data, you may actually get a valid RGB pixel with black color, and then you'll have to decide whether this is the color obtained from the sensor or that it received it during alignment as RGB couldn't find its correspondence in the depth frame.

BadMachine, 2019-12-09:
The frames:
[image]

q: "The alignment takes place before sending the frame to pose estimator"
a: yes, ofc

q: "The pose estimator extract the tracking features from the RGB image only."
a: yes, it works only with RGB channel

q: "The 3d coordinates are extracted from the original depth image."
a: if u meant that giving depth value is from depth frame, then yes  
``` c++
float pixel_distance_in_meters = depth.get_distance(toSend.captured[i.key()].x, toSend.captured[i.key()].y);
rs2_deproject_pixel_to_point(planarPoint3d, &inrist, pix, pixel_distance_in_meters);
```

ev-mp, 2019-12-09:
The skeleton tracking for the right hand seems to locate the palm outside of the textured (aligned) area and right into the black area. See the highlighted red dot below. To the very least - it is on the border line so it makes sense that there would be no corresponding depth data.

[image]

Once you manage to configure the estimator to move the palm's location away from the no-valid/boundary data towards the `center of the mass` of the right hand where it actually belongs (the convex colored shape that represents it), then IMHO you'll manage to get a consistent depth data in a reliable manner.

BadMachine, 2019-12-09:
@ev-mp 
Thank you for clarifying, drew attention to my problem and flattering reviews about my app architecture, really appreciate it!
Seems i should close the issue, cheers!

closed 2019-12-09 by BadMachine

ev-mp, 2019-12-09:
I just realized that in order to enhance depth<->color alignment required by tracking you also must ensure the temporal sync between the color and depth frames capture times is within very strict boundaries. 
Using 15 FPS means that potentially there could be at least 60msec latency between color and depth (internal sync is off), so when tracking body in motion the delay affect the correspondence negatively. The ways to alleviate this is to increase FPS, set the RGB sensor's `Exposure_priority` control off (#5290) and verify that the actual frame rates are 30+. Enabling metadata is also recommended as it will allow to use HW rather than Host-generated timestamps ([installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows)).
I would suggest to experiment with high FPS values, preferably 60 FPS at the cost of lower resolution

BadMachine, 2019-12-09:
@ev-mp thanks for advice
but 15 fps - not realsense stream determinated value, it`s quite max value while using neural network for pose estimator, unfortunately)

BadMachine, 2019-12-09:
@ev-mp 
I think i found right way:
Using chain of post-processing pilters, like this:

```c++
frames = frames.apply_filter(dec_filter).apply_filter(spat_filter).apply_filter(temp_filter).apply_filter(hole_filter);
```
[image]

ev-mp, 2019-12-10:
@BadMachine , that's correct - post-processing shall be performed before alignment to avoid smearing of the generated artifacts. See a similar reply in [another thread](https://github.com/IntelRealSense/librealsense/issues/5385#issuecomment-563402118)