realsenseai/librealsense #5405: ImportError: No module named rospy

tmichals, 2019-12-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T235 | 
| Firmware Version                   | RealSense ROS | 
| Operating System & Version |  Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  Linux 4.9.140-tegra aarch64                                    | 
| Platform                                 | NVIDIA Jetson|
| SDK Version                            |    JetPack 3.3.1           | 

I am following the steps from below...
https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

I have both my T265 and D435 camera's running as per step 2:   

On Step 3: I get the following error  "ImportError: No module named rospy" ?  Any idea's?
lcs@nano:~/catkin_ws/src$ cd ..
lcs@nano:~/catkin_ws$ python src/realsense-ros/realsense2_camera/scripts/set_cams_transforms.py cam_1_link cam_2_link 0.7 0.6 0 -90 0 0
Traceback (most recent call last):
  File "src/realsense-ros/realsense2_camera/scripts/set_cams_transforms.py", line 3, in <module>
    import tf
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 39, in <module>
    from .buffer_interface import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/buffer_interface.py", line 32, in <module>
    import roslib; roslib.load_manifest('tf2_ros')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 62, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 93, in _generate_python_path
    m = rospack.get_manifest(pkg)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 414, in parse_manifest_file
    _static_rosdep_view = init_rospack_interface()
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 133, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 640, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 536, in load_cached_sources_list
    with open(cache_index, 'r') as f:
IOError: [Errno 13] Permission denied: '/home/lcs/.ros/rosdep/sources.cache/index'
lcs@nano:~/catkin_ws$ sudo python src/realsense-ros/realsense2_camera/scripts/set_cams_transforms.py cam_1_link cam_2_link 0.7 0.6 0 -90 0 0
[sudo] password for lcs:
Traceback (most recent call last):
  File "src/realsense-ros/realsense2_camera/scripts/set_cams_transforms.py", line 1, in <module>
    import rospy
ImportError: No module named rospy
lcs@nano:~/catkin_ws$ ImportError: No module named rospy
-bash: ImportError:: command not found

dorodnic, 2019-12-10:
Well, you need to the rospy package
Have you tried `sudo apt-get install -y python-rospy`?
Not sure if it is available for Jetson, but this would be general ROS question, not directly related to RealSense.

tmichals, 2019-12-12:
Thank you for your help... I tried and I got a defiant reply...__________________________________lcs@nano:~/catkin_ws$ sudo apt-get install -y python-rospy[sudo] password for lcs: Reading package lists... DoneBuilding dependency tree       Reading state information... DoneSome packages could not be installed. This may mean that you haverequested an impossible situation or if you are using the unstabledistribution that some required packages have not yet been createdor been moved out of Incoming.The following information may help resolve the situation:
The following packages have unmet dependencies: python-rospy : Depends: python-roslib but it is not going to be installedE: Unable to correct problems, you have held broken packages.lcs@nano:~/catkin_ws$

tmichals, 2019-12-23:
Please close... this is probably an issue but I have gotten around needing it.

closed 2019-12-23 by tmichals