realsenseai/librealsense #5406: update sdk version question

TouchDeeper, 2019-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300| 
| Firmware Version                   | 3.15.0.0 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-72-generic                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.29.0                          | 
| Language                            |  c++                         | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to update realsense sdk from 2.18.1 to 2.29.0.  I have `sudo make uninstall` in 2.18.1.

When I process `./scripts/patch-realsense-ubuntu-lts.sh`,  output below:
```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-hwe folder

Applying patches for xenial-hwe line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
```
I hit enter two times. This should because I have install patch when I install 2.18.1.

After I install the 2.29.0, I tried to test `realsense-viewer`,  the ouput of  `realsen-viewer` is below:
```
 09/12 11:10:11,150 WARNING [140289839146752] (backend-v4l2.cpp:1148) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 09/12 11:10:11,150 WARNING [140289839146752] (backend-v4l2.cpp:1148) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 09/12 11:10:16,203 WARNING [140289839146752] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 09/12 11:10:16,212 WARNING [140289839146752] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 09/12 11:10:16,212 ERROR [140289839146752] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 09/12 11:10:16,216 WARNING [140289839146752] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Enable / disable auto-exposure
 09/12 11:11:13,773 WARNING [140289729619712] (sr300.cpp:358) UVC metadata payloads are not available for stream 5a3136201513174560. Please refer to installation chapter for details.
 09/12 11:11:14,362 WARNING [140289708640000] (sr300.cpp:358) UVC metadata payloads are not available for stream 595559561498765654. Please refer to installation chapter for details.
```
The warning of `UVC metadata payloads are not available` may be the problem caused by patch kernal.

When I tried to run ros node `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`, error occured:
```
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
[... 109 lines cut ...]
[ INFO] [1575860845.562063192]: Setting Dynamic reconfig parameters.
[ INFO] [1575860845.606874738]: Done Setting Dynamic reconfig parameters.
 09/12 11:07:25,607 WARNING [139691507070720] (backend-v4l2.cpp:1148) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 09/12 11:07:25,607 WARNING [139691507070720] (backend-v4l2.cpp:1148) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
[ INFO] [1575860845.625755229]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1575860845.625910424]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1575860845.627022328]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1575860845.627044863]: setupPublishers...
[ INFO] [1575860845.630962237]: Expected frequency for depth = 30.00000
[ INFO] [1575860845.639489333]: Expected frequency for infra1 = 30.00000
[ INFO] [1575860845.641058470]: Expected frequency for color = 30.00000
[ INFO] [1575860845.643548891]: setupStreams...
[ INFO] [1575860845.643885718]: insert Depth to Coded-Light Depth Sensor
[ INFO] [1575860845.643911383]: insert Color to RGB Camera
[ INFO] [1575860845.643922102]: insert Infrared to Coded-Light Depth Sensor
[ INFO] [1575860845.648392546]: SELECTED BASE:Depth, 0
[ INFO] [1575860845.653646791]: RealSense Node Is Up!
 09/12 11:07:26,010 WARNING [139691473499904] (sr300.cpp:358) UVC metadata payloads are not available for stream 494e5a491229871689. Please refer to installation chapter for details.
[ WARN] [1575860846.013629230]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 09/12 11:07:26,223 WARNING [139691371808512] (sr300.cpp:358) UVC metadata payloads are not available for stream 595559561498765654. Please refer to installation chapter for details.
 09/12 11:07:26,225 WARNING [139691341784832] (types.cpp:48) Region-of-interest is not implemented for this device!
[ERROR] [1575860846.238346599]: Region-of-interest is not implemented for this device!
```
There still has the error `UVC metadata payloads are not available `. and new error `Region-of-interest is not implemented for this device!`  occured?
Could you please provide some advice? Thanks a lot!

TouchDeeper, 2019-12-10:
anyone help~

dorodnic, 2019-12-10:
Hi @TouchDeeper 
Will check. Its the combination of SDK, ROS-RealSense and SR300 firmware version that makes the wrapper think it should support RGB auto-exposure ROI, when in fact it should not.

TouchDeeper, 2019-12-11:
@dorodnic  Thank you for reply.

Is the realsense-viewer's warning `UVC metadata payloads are not available for stream 5a3136201513174560. Please refer to installation chapter for details.` also a bug? If not, how can I fix it? Thanks a lot!

ipa-mah, 2019-12-16:
I also have a problem with librealsense and realsense-ros
https://github.com/IntelRealSense/realsense-ros/issues/1017
https://github.com/IntelRealSense/librealsense/issues/5233

RealSenseCustomerSupport, 2020-07-02:
----------------------------------------------

We will fix the issue. But the schedule is not available yet.

doronhi, 2021-09-01:
In a very long delay, using librealsense2 version 2.48.0 and realsense-ros version 2.3.1 and an SR305 I can no longer reproduce the `Region-of-interest is not implemented for this device!` error.

closed 2021-09-08 by RealSenseSupport