realsenseai/librealsense #5407: Face features to be selected - Dlib

RajkiranVeldur, 2019-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.30.0 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-36-generic                                        | 
| Platform                                 | PC - NVIDIA:390.116  |
| SDK Version                            |  NA                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Please help us with the landmarks you are selecting
Can you please tell me what are the landmarks you are considering for the facial features(Ear, Nose, Eye, Chin and Mouth) with respect to the image below?

As we are trying to write a Python version of your C++ algorithm, we are experiencing issues with improvising accuracy. 

Hence, we would like to ensure that we are taking the correct landmarks for calculating depth.

RajkiranVeldur, 2019-12-09:
[image]

MartyG-RealSense, 2019-12-09:
Reading the example program's instructions, it looks as though it is using the pre-made IMAVIS "300-W" 68-point training data file.  300-W refers to " 300 faces In-the-wild challenge".

A research paper was written about it, and the landmarks are illustrated in more detail on pages 10 and 11 of the paper.

https://ibug.doc.ic.ac.uk/media/uploads/documents/sagonas_2016_imavis.pdf

RajkiranVeldur, 2019-12-09:
@MartyG-RealSense thanks for the response. However, I am looking for the exact points that you people are considering in the markup_68.h(librealsense/wrappers/dlib/face/markup_68.h).

MartyG-RealSense, 2019-12-09:
Looking at the text description of the points:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/dlib/face/markup_68.h

It looks as though the list of points are described in order of the ascending direction of the numbers.  

So for example, for points 1-17, it suggests the right ear (1) as the start point.  

**// Starting with right ear, the jaw [1-17]
    RIGHT_EAR, JAW_FROM = RIGHT_EAR, RIGHT_JAW_FROM = RIGHT_EAR,
    RIGHT_1, RIGHT_2, RIGHT_3, RIGHT_4, RIGHT_5, RIGHT_6, RIGHT_7, RIGHT_JAW_TO = RIGHT_7,
    CHIN, CHIN_FROM = CHIN - 1, CHIN_TO = CHIN + 1,
    LEFT_7 = CHIN + 1, LEFT_JAW_FROM = LEFT_7, LEFT_6, LEFT_5, LEFT_4, LEFT_3, LEFT_2, LEFT_1,
    LEFT_EAR, LEFT_JAW_TO = LEFT_EAR, JAW_TO = LEFT_EAR,**

So if you read through the list of descriptions for the points in this section, starting at Point 1 (the right ear), it takes you down the right side of the face to the chin and then up the left side of the face to end at the left ear at point 17.

RajkiranVeldur, 2019-12-09:
I understand this file but since I was using in Python with MTCNN, I observed a drop in the accuracy. So to be sure about the landmarks, can you please help me to identify the landmarks of these points according to the image above:

1. markup_68::RIGHT_EAR, markup_68::RIGHT_1
2. markup_68::LEFT_1, markup_68::LEFT_EAR
3. markup_68::CHIN_FROM, markup_68::CHIN_TO
4. markup_68::NOSE_RIDGE_2, markup_68::NOSE_TIP
5. markup_68::RIGHT_EYE_FROM, markup_68::RIGHT_EYE_TO
6. markup_68::LEFT_EYE_FROM, markup_68::LEFT_EYE_TO
7. markup_68::MOUTH_OUTER_FROM, markup_68::MOUTH_INNER_TO

Please let me know. I will be really thankful. 

@maloel  :  Please add comments if you have any input regarding the same.

MartyG-RealSense, 2019-12-09:
I studied the chart very carefully and compared the points to the text descriptions.  I would rather let one of the Intel team answer this one definitively rather than make a guess and advise you badly.  I do apologize.

maloel, 2019-12-09:
Hi @RajkiranVeldur 
The points in markup_68 are 0-based, while the points in the image are 1-based. Keep that in mind.

I.e., RIGHT_EAR is point 0 in the sample, '1' in the image.
Likewise, LEFT_EAR is '17' in the image but point 16 in the code.

Hope that clarifies it?

RajkiranVeldur, 2019-12-10:
@maloel thank you for the reply. I got that indices are as per the Pythons index count, i.e., starting from zero. However,  to be sure can you please help me with every facial landmark point, I have asked in the above comment.

dorodnic, 2019-12-10:
@RajkiranVeldur - expanding what is provided in the C++ example currently goes beyond the scope of this project. We are trying to focus on RealSense specific aspects in this forum. 
If you figure out how to port this example to python you are encouraged to publish your findings. 
Otherwise, we will revisit adding equivalent python example in the future.

RajkiranVeldur, 2019-12-11:
@maloel @dorodnic I get the reference of the indices. I'm just confused with the exact value of indices, that you are using in C++ code for the following landmarks in the image given above. 

Please tell the indices for all these given landmarks below, I will be really thankful.

1. markup_68::RIGHT_EAR, markup_68::RIGHT_1
2. markup_68::LEFT_1, markup_68::LEFT_EAR
3. markup_68::CHIN_FROM, markup_68::CHIN_TO
4. markup_68::NOSE_RIDGE_2, markup_68::NOSE_TIP
5. markup_68::RIGHT_EYE_FROM, markup_68::RIGHT_EYE_TO
6. markup_68::LEFT_EYE_FROM, markup_68::LEFT_EYE_TO
7. markup_68::MOUTH_OUTER_FROM, markup_68::MOUTH_INNER_TO

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

Hi @RajkiranVeldur, Like dorodnic commented, this discussion is beyond the scope of this project. Do you have further questions?
If we don't hear from you in 7 days, this issues will be closed. Thanks.

closed 2020-02-11 by RealSenseCustomerSupport