realsenseai/librealsense #5409: Documentation of advanced controls on Realsense viewer

vishnukumarts, 2019-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.072)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to understand what are all controls available in the Realsense viewer under Advanced controls section.  

I already referred the below pages,
https://www.intelrealsense.com/wp-content/uploads/2019/11/BKMs_Tuning_RealSense_D4xx_Cam.pdf

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

But these pages still did not provide complete data.

MartyG-RealSense, 2019-12-09:
Intel did not create complete documentation for all of the Advanced Mode controls.   This was because these functions interact with each other in complex ways, and so Intel chose to control them with machine-learning algorithms instead.  RealSense users are welcome to experiment with the settings and see how they affect the output though.

vishnukumarts, 2019-12-09:
Testing by two unknowns(input & output) terms won't help the experiment process.
It would be helpful, you guys share the full form of the following abbreviations, of the controls.  So that I can understand the input and expected output.

Thanks.

MartyG-RealSense, 2019-12-09:
I understand, and agree that a list of full explanations for the Advanced Mode abbreviations would be very useful - it is something I have had need of myself in the past.  I am not a member of the Intel RealSense team though, so I will have to defer to them regarding whether it is possible to publish such a list at some point.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Thank you @vishnukumarts

We will keep this in mind and noted as we look at our documentation improvements in the future.

If there is nothing else necessary, then we can close out this thread.

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport