realsenseai/librealsense #5410: What's the difference between these two methods to get depth distance

zhangzhe1103, 2019-12-09:
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | RealSense Viewer --> ２.１７ | 
| Operating System & Version |   Linux (Ubuntu 16   | 
| Kernel Version (Linux Only)    |  4.15                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson  |
| SDK Version                            |  { ２.１７ }                          | 
| Language                            |  python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello! 
What's the difference between these two methods to get depth distance?
the first method by function get_distance
the second method by pointcloud
`import cv2
import numpy as np
import pyrealsense2 as rs
class intel_video_capture:
    def __init__(self):
        use_cuda = 1
        # Declare pointcloud object, for calculating pointclouds and texture mappings
        self.pc = rs.pointcloud()
        # We want the points object to be persistent so we can display the last cloud when a frame drops
        # points = rs.points()
        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 15)
        # Start streaming
        self.pipeline.start(config)
        # Create an align object
        # rs.align allows us to perform alignment of depth frames to others frames
        # The "align_to" is the stream type to which we plan to align depth frames.
        align_to = rs.stream.color
        self.align = rs.align(align_to)

    def frame_reader(self):
        # Wait for a coherent pair of frames: depth and color
        frames = self.pipeline.wait_for_frames()
        # Align the depth frame to color frame
        aligned_frames = self.align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()

        # the first method by function get_distance
        dist_to_center = depth_frame.get_distance(320, 240)

        width = depth_frame.get_width()
        height = depth_frame.get_height()
        print('wh:', width, height)
        print('get_distance:', dist_to_center)
        color_frame = aligned_frames.get_color_frame()
        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        # pointcloud #depth calculation
        self.pc.map_to(color_frame)
        points = self.pc.calculate(depth_frame)
        vtx = np.asanyarray(points.get_vertices())

        # the second method by pointcloud
        vtx = np.reshape(vtx, (480, 640, -1))  # 3d depth info
        print('vtx:', vtx[320][240][2])

        print('*' * 30)
        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        return color_image, depth_colormap, vtx

    def frame_stop(self):
        self.pipeline.stop()`

ev-mp, 2019-12-09:
@zhangzhe1103 , there should be no difference - those two methods are functionally equivalent.
There are potentially inconsistencies on the level of numeric precision due different implementation pipes employed by the processing blocks (CPU/SSE/GLSL/CUDA). 
Performance-wise there is a difference, as the first method calculates the depth value for a specific pixel while the second method calculates [xyz] vertices for all frame pixels.
P.S: I added minor formatting change for readability

zhangzhe1103, 2019-12-10:
> @zhangzhe1103 , there should be no difference - those two methods are functionally equivalent.
> There are potentially inconsistencies on the level of numeric precision due different implementation [...]

Thank you very much, and yesterday I found that there are some problems with the format but I don’t know how to modify it. It looks much more comfortable after changing

closed 2019-12-10 by zhangzhe1103