realsenseai/librealsense #5412: Raspberry Pi 4: high CPU utilization freezes color image

SirDifferential, 2019-12-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version | Raspbian 10  | 
| Kernel Version (Linux Only)    | 4.19.75-v7l+ | 
| Platform                                 | Raspberry Pi 4.  |
| SDK Version                            |  Latest master, 38c4fdaaa20aef28fa40365bfb201913673a5ac0 | 
| Language                            |  C++ | 
| Segment			|  Streaming | 

### Issue Description

When running realsense viewer with D435 with 1280x720 depth at 30 FPS and color 1920x1080 at 30 FPS on a Raspbery Pi 4, the color image freezes if CPU utilization is high. The color image works fine CPU utilization is not too high. This happens when building with `-DFORCE_RSUSB_BACKEND=ON`.

### To reproduce

* Start realsense viewer with 1280x720@30 depth and 1920x1080@30 color
* In another terminal run a CPU stress tool (available in standard repos): `stress -c 4`

### Notes

* Observe `/var/log/kern.log`
* Before the freeze happens, a few errors like this are printed:

```
Dec  9 11:27:08 raspberrypi kernel: [  813.960741] usb 2-1: Process 837 (realsense-viewe) called USBDEVFS_CLEAR_HALT for active endpoint 0x84
```

* When these errors happen, the color image displays the auto exposure settings being changed (the image appears more brown or brighter before normalizing again)
* I attempted to test only the exposure settings by not stressing the CPU and tricking the camera into auto exposure changes. The color image did not freeze.
* When the freeze happens, this error is printed:

```
Dec  9 11:28:05 raspberrypi kernel: [  870.938769] usb 2-1: Process 837 (realsense-viewe) called USBDEVFS_CLEAR_HALT for active endpoint 0x84
Dec  9 11:28:05 raspberrypi kernel: [  870.938989] usb 2-1: EP not empty, refuse reset
Dec  9 11:28:05 raspberrypi kernel: [  870.945758] xhci_hcd 0000:01:00.0: Can't enqueue URB while manually clearing toggle
Dec  9 11:28:05 raspberrypi kernel: [  870.946082] usb 2-1: usbfs: usb_submit_urb returned -22
Dec  9 11:28:05 raspberrypi kernel: [  870.969731] xhci_hcd 0000:01:00.0: Can't enqueue URB while manually clearing toggle
Dec  9 11:28:05 raspberrypi kernel: [  870.970108] usb 2-1: usbfs: usb_submit_urb returned -22
```

* The camera works again after restarting the dead color feed. Depth feed works without interruption.
* The color image does not appear to freeze at other resolutions or framerates. 60 FPS was not tested. Color at 1920x1080@15 seems to work.
* With high CPU utilization, the color image also appears to get delayed several seconds behind the depth image. Perhaps this suggests at some sort of buffering issue that eventually produces this problem.
* I have a video showing this issue [here](https://www.youtube.com/watch?v=aN42JTIvnG0&feature=youtu.be). The freeze happens at 05:28.

dorodnic, 2019-12-10:
Hi @SirDifferential, good to hear from you again.
Investigating high CPU utilization report on rsusb backend is high on our priority list, sorry for some delay, there are a lot of different things happening with RealSense currently.
We are now testing the Pi more extensively and so far we have seen it perform very well.  
Its probably stupid question, but did you check the `-DCMAKE_BUILD_TYPE=Release` flag? (once we finished Debians for Jetson family, RPi4 is next in line, so you will not have to deal with manual builds for long)
We are also working on switching to libyuv in the SDK, as it would give superior experience to arm users. 
I can't quite figure out from the screen capture - what is the total CPU utilization across cores you are aiming at? USB-Bulk device will always struggle when approaching 100%, regardless of backend and platform. We will try to reproduce the AE behavior, but at some point all queues involved (in user-space or kernel) will fill up and you are likely to see frame drops (latency should be always bounded, will check whats going on with that)

SirDifferential, 2019-12-11:
> Its probably stupid question, but did you check the -DCMAKE_BUILD_TYPE=Release flag?

Not a stupid question. I assumed it was default and didn't specify it.

> We are also working on switching to libyuv in the SDK, as it would give superior experience to arm users.

Sounds good.

> I can't quite figure out from the screen capture - what is the total CPU utilization across cores you are aiming at?

I first observed this in our actual application where we are pretty much driving the RPI4 CPU to its limits. To reproduce this problem I just stressed the CPU to its maximum capacity (4 threads executing at 100%) and ran the realsense viewer.

> USB-Bulk device will always struggle when approaching 100%, regardless of backend and platform. We will try to reproduce the AE behavior, but at some point all queues involved (in user-space or [...]

I figured the frame drops would happen, but the stream freezing in full seems like unintended behavior.

kafan1986, 2019-12-14:
> Hi @SirDifferential, good to hear from you again.
> Investigating high CPU utilization report on rsusb backend is high on our priority list, sorry for some delay, there are a lot of different things [...]

Hmm. Same on Android platform on arm device. I believe one does not suffer from RSUSB issues on desktop environment, given the modern single thread performance is much faster on modern desktop processor, thus it avoids issues faced by comparatively lower processing device like SBC.

Although the platform is designed for a producer consumer platform and on a lower processing environment, and at worst one should only see dropped frame but here the problem seem to be complete stall of color pipeline.

On Android the system suffers from memory management issues, which happens when the USB Request fails and queue fills up and even the cancel request to free up the memory fails.

`usb_request_cancel returned error, endpoint: 132, error: Invalid argument, number: 22`

Apart from reducing processing on RSUSB, I believe the isochronous data transfer is not implemented. It might be worth a try to see if it helps. I found a project that did the same for android platform. https://github.com/Peter-St/Android-UVC-Camera https://play.google.com/store/apps/details?id=humer.uvc_camera&hl=en

Another approach can be to assign a dedicated core for data transfer. This can be achieved via CPU set on linux and Android on version 7 and above. One needs to modify the native wrapper. An example for android platform is shown in this example
https://github.com/paulnavin/Apollo/blob/4210caf76e2f8f5212af7558d9d59dff75ef3500/app/src/main/cpp/oboe/samples/MegaDrone/src/main/cpp/AudioEngine.cpp

kafan1986, 2019-12-16:
I am running streaming depth and colour frame on android for last 36+ hours at 720p 15fps without any crash. I do get some messages like below every few minutes but it has yet to give any additional critical error like OOM.

`W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32`

Basically currently my code is just getting the depth and colour frame with just some post processing filter and nothing else. Top command shows the CPU usage to be around 40-45%. Earlier I was also drawing the eachb colour frame to UI and usually the app would crash in minutes.

It means although the data transfer is the culprit and the memory leak is not in the SDK itself but in the way how the RSUSB is implemented. When the CPU is not stressed then RSUSB can work fine but when it is stressed the USB request fail and then I believe the method of cancelling USB request, that should cancel the filled URB queues held with the linux kernel is not working properly releasing the queue and thus causes Out of memory error. 

As expected in such case hardware reset or even custom USB controller reset is not able to recover and only option is to reboot the entire system.

SirDifferential, 2019-12-17:
I'm also verifying that 1920x1080 15fps has not frozen the color feed on my three RPI4s after 4 days of running. It seems to be a problem that only happens when you ramp up the feed to 1920x1080 at 30 fps.

kafan1986, 2019-12-17:
@SirDifferential 
You are running on RPI4 with Raspbian 10 thats why you are atleast able to do 1080p on 15 fps. Perhaps the RSUSB implementation is even worse on Android. Here on android even with Odroid N2 which around almost twice as powerful as Rpi4 both single core and multicore, I am not able to do anything with 1080p.

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@SirDifferential We do see some issues with 1920x1080@30fps on some android platforms. Is it possible for you to workaround it with lower resolution or frame rate before we get the solution? Thanks!

SirDifferential, 2020-03-19:
Yes, our project can work with a lower framerate for now. I'm mostly worried if the underlying problem could perhaps manifest with lower framerates as well.

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@SirDifferential We did long duration stability testing with Depth and Color streaming at 640x480@30fps on RK3399 android system and didn't get issues after 2 weeks running. Though we didn't get chance to try higher resolution with 15fps for such longer duration testing, we are sure about that using lower resolution or frame rate can get more robust stability on some android platform such as Raspberry or RK3399.

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@SirDifferentia Any other questions about this? Thanks!

SirDifferential, 2020-03-26:
No, we can close this. Thanks!

closed 2020-03-26 by SirDifferential