realsenseai/librealsense #5414: depth factor for saving Depth image (one channel 16bit Gray scale)

Dennis2Choi, 2019-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / **D400** } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {**Win (10)** / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | **PC**/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/**python**/unity }                          | 
| Segment			|  {**Robot**/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm working on the depth map extraction from bag file.

need special format for depth map (16bit grayscale depth map image with 640x480 resolution)

usually the values from depth map * "1,000" = real distance in meters
(so , let's call depth factor 1,000)

but after my code, I think it would be changed "10,000" 

Is there anyone who knows about depth factor or Do I need some modification for that?

Here is my code 
==============================================================
#####################################################
##   Extract depth PNG image from bag file         ##
#####################################################

# First import library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2
# Import argparse for command-line options
import argparse
# Import os.path for file path manipulation
import os.path

#####################################################
## Save path & total frame number
path = "c://temp/"
totalnum = 1
Frame_names = "1"

# Create object for parsing command-line options
parser = argparse.ArgumentParser(description="Read recorded bag file and display depth stream in black to white colormap.\
                                Remember to change the stream resolution, fps and format to match the recorded.")
# Add argument which takes path to a bag file as an input
parser.add_argument("-i", "--input", type=str, help="Path to the bag file")
# Parse the command line arguments to an object
args = parser.parse_args()
# Safety if no parameter have been given
if not args.input:
    print("No input paramater have been given.")
    print("For help type --help")
    exit()
# Check if the given file have bag extension
if os.path.splitext(args.input)[1] != ".bag":
    print("The given file is not of correct file format.")
    print("Only .bag files are accepted")
    exit()

def read_bag_save_png():
    # Create pipeline
    pipeline = rs.pipeline()

    # Create a config object
    config = rs.config()
    # Tell config that we will use a recorded device from filem to be used by the pipeline through playback.
    rs.config.enable_device_from_file(config, args.input)
    # Configure the pipeline to stream the depth stream
    config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

    # Start streaming from file
    pipeline.start(config)

    # Create opencv window to render image in
    cv2.namedWindow("Depth Stream", cv2.WINDOW_AUTOSIZE)
    
    # Create colorizer object
    colorizer = rs.colorizer();
        
    # Set to colorizer to White-to-black(2) or Black-to-white(3)
    # colorizer.set_option(rs.option.histogram_equalization_enabled,0)
    colorizer.set_option(rs.option.color_scheme, 3)

    # Streaming loop
    try:
        i = 1
        while True:
            # Get frameset of depth
            frames = pipeline.wait_for_frames()

            # Get depth frame
            depth_frame = frames.get_depth_frame()

            # Colorize depth frame to jet colormap
            depth_color_frame = colorizer.colorize(depth_frame)

            # Convert depth_frame to numpy array to render image in opencv
            depth_color_image = np.asanyarray(depth_color_frame.get_data())

            # Convert depth_color_image [R,G,B, 3 channel] to depth_gray_image [1 channel]
            depth_gray_image = cv2.cvtColor(depth_color_image, cv2.COLOR_RGB2GRAY)
        
            # Convert 16bit data
            detph_gray_16bit = np.array(depth_gray_image, dtype=np.uint16)
            detph_gray_16bit *= 256

            # Resize Depth image to 640x480
            resized = cv2.resize(detph_gray_16bit, (640,480), interpolation = cv2.INTER_AREA)
        
            print("Saving frame:", i)
            # Save the file 
            status = cv2.imwrite(path +  "/" + Frame_names +"_Depth_" + str(i).zfill(6) + ".png", resized)
            i += 1

            # if i == totalnum, escape exit program
            if i == totalnum+1:
                cv2.destroyAllWindows()
                break

            # Render image in opencv window
            cv2.imshow("Depth Stream", resized)
        
            key = cv2.waitKey(1)
            # if pressed escape exit program
            if key == 27:
                cv2.destroyAllWindows()
                break
    finally:
        pipeline.stop()    

if __name__ == "__main__":
    read_bag_save_png()

ev-mp, 2019-12-09:
@Dennis2Choi , there is an option called `RS2_OPTION_DEPTH_UNITS` that provides the depth scale factor. During depth streaming its value can be altered, and by doing so you also modify the raw depth data. You can run `rs-enumerate-devices -o` to check the option's range, step and default attributes.
The code above does not set/query this option, so it is not clear  what do you mean by 
>usually the values from depth map * "1,000" = real distance in meters
(so , let's call depth factor 1,000)
but after my code, I think it would be changed "10,000"

Do you mean that the rosbag was recorded with the depth unit set to a non-default value ?
Please elaborate

Dennis2Choi, 2019-12-09:
Thank you @ev-mp 
I will try and give you feedback

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @Dennis2Choi,

Has the comment provided by @ev-mp helped to find an answer to your question?Do you need any further assistance with this matter?

Thank you!

closed 2019-12-23 by RealSenseCustomerSupport