realsenseai/librealsense #5416: pointcloud exception using recording in unity

shenberg, 2019-12-09:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.22.0 }                          | 
| Language                            | unity                          | 

### Issue Description

I can't create a point cloud using a file recording I just created using realsense viewer. Same settings work great for live input but it's difficult to develop this way.

I created recording with 640x480 for both depth and RGB. Output format for RGB is rgba8, depth is Z16. When starting the scene, I get an error saying there's an issue in the point cloud processing block. I create a log file as detailed in [issue 1896](https://github.com/IntelRealSense/librealsense/issues/1896) and each frame, there's one ERROR row and one WARNING preceding it generated:
```
 09/12 11:31:10,901 WARNING [21948] (types.cpp:48) Device does not support option Depth Units!
 09/12 11:31:10,901 ERROR [21948] (synthetic-stream.cpp:44) Exception was thrown during user processing callback!
```

Full log attached.
[debug.txt](https://github.com/IntelRealSense/librealsense/files/3941460/debug.txt)

Unity stack trace:
```
InvalidOperationException: Error while running PointCloud processing block, check the log for info
Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.ProcessingBlock.Process (Intel.RealSense.Frame original) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsPointCloud.Process (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource frameSource) (at Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsPointCloud.cs:59)
RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:100)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:120)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsProcessingPipe:OnFrame(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:69)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockFrameCallback(IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_synthetic_frame_ready(IntPtr, IntPtr, Object&)
Intel.RealSense.FrameSource:FrameReady(Frame)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:113)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:149)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:161)
System.Threading.ThreadHelper:ThreadStart()
```

wodndb, 2019-12-19:
I understand that you want to record point cloud frames using Rs Device prefab. Error is occurred in RsProcessingPipe, and Something wrong about depth units. I think that processing pipe setting or RsDevice Setting are wrong.
Could you share your processing pipe setting and RsDevice prefab settings which can be show in inspector?

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@shenberg Sorry for late response! Did you still have this issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@shenberg Could you please update? Thanks!

shenberg, 2020-04-22:
I'm sorry, this was irrelevant by the time a response was received. It seems to me that there is some setting I missed where the recording clashes with the prefab. I based myself on Keijiro Takahashi's [Rsvfx repository](https://github.com/keijiro/Rsvfx), try any of the example scenes there.

RealSenseCustomerSupport, 2020-04-23:
----------------------------------------------

@shenberg So no other issues or questions about this ticket from your side, right? Thanks for you reply!

RealSenseCustomerSupport, 2020-05-06:
----------------------------------------------

@shenberg Close this if no other questions about it. Thanks!

closed 2020-05-06 by RealSenseCustomerSupport