realsenseai/librealsense #5418: Android colour stream crashes with Out of memory issue with 1080p and 720p config

kafan1986, 2019-12-10:
Required Info |  
-- | --
Camera Model | { D415}
Firmware Version | {05.11.15.00}
Operating System & Version | {Android}
Platform | {Arm-Odroid}
SDK Version | { 2.30.0 }
Language | {C++/Java }
Segment | {others }

I am using D415 on android and running a Colour stream at (1920x1080x15fps) and Depth stream at (1280x720x15fps and decimation filter with decimation magnitude of 2). The colour stream eventually gets stuck with Out of memory error and depth continues to work. Following approach to correct the issue does not work.
a) Creating a new pipeline
b) Calling Hardware Reset
c) USB hub reset

Only rebooting the device works.

If the colour stream is set at 720px15fps, even then the error arises.

If the colour stream is set at 480px15fps then although the below error is printed but device continues to work for the few hours that I have tested till now. I will update it after testing for longer duration.

480p stream error:

```
2019-12-10 08:19:59.461 4712-4919/com.synapticsense.retailapp I/chatty: uid=10061(com.synapticsense.retailapp) Thread-3 identical 12 lines
2019-12-10 08:19:59.521 4712-4919/com.synapticsense.retailapp W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32
2019-12-10 08:24:15.435 4712-4919/com.synapticsense.retailapp W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32
```

I have posted in details other logs on following post
POST:
https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-561935062
THREAD:
https://github.com/IntelRealSense/librealsense/issues/4091

Any guesses what causes "out of memory" issue, is it due to USB driver or D415 ISP/firmware related issue? 

I have noticed there is already an issue: **(DSO-13539) - [Android] Camera disconnected after streaming some duration with Android Camera Sample** related to this but there is no resolution till now.

I am not great in C++ but would love to help anyone debug and solve the root cause.

kafan1986, 2019-12-16:
I am running streaming depth and colour frame on android for last 36+ hours at 720p 15fps without any crash. I do get some messages like below every few minutes but it has yet to give any additional critical error like OOM.

`W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32`

Basically currently my code is just getting the depth and colour frame with just some post processing filter and nothing else. Top command shows the CPU usage to be around 40-45%. Earlier I was also drawing the eachb colour frame to UI and usually the app would crash in minutes.

It means although the data transfer is the culprit and the memory leak is not in the SDK itself but in the way how the RSUSB is implemented. When the CPU is not stressed then RSUSB can work fine but when it is stressed the USB request fail and then I believe the method of cancelling USB request, that should cancel the filled URB queues held with the linux kernel is not working properly releasing the queue and thus causes Out of memory error. 

As expected in such case hardware reset or even custom USB controller reset is not able to recover and only option is to reboot the entire system.

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@kafan1986 Did you get any chance to try the latest SDK? We have several fixes after v2.30 including memory leakage fix.  However, we do see some stability issues with HD or FHD resolution at 30fps on some android platforms such as Raspberry, Odroid. Is it possible for you to workaround with lower resolution before we get solution? Thanks!

kafan1986, 2020-03-19:
I am currently using the latest SDK at 720p and 15 fps, to gain maximum stability, which then runs for atleast 5-6 hours. I wish to operate operate at 1080p and 30 fps but the stability of the system goes nose diving then and system starts giving error in less than an hour's time.

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@kafan1986 Is it possible for you to lower the resolution to 640x480 to see if better? Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@kafan1986 Could you please update? Thanks!

kafan1986, 2020-04-08:
@RealSense-Customer-Engineering 640p resolution of RGB camera would not be sufficient for my use case.

RealSenseCustomerSupport, 2020-05-29:
----------------------------------------------

Tracked by DSO-14875

RealSenseSupport, 2021-05-12:
@kafan1986 We have some Android stability enhancement after SDK v2.39.  https://github.com/IntelRealSense/librealsense/pull/7205

Could you please try the latest SDK v2.45 to see if you still get this issue? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-05-19:
@kafan1986 Did you get any chance to try the latest SDK v2.45? Please note that the issue will be closed if we don't hear from you for another 7 days. Thanks!

RealSenseSupport, 2021-05-25:
@kafan1986 Sorry that we don't hear from you. Will close the ticket at this point. Please feel free to create another new ticket if you still have issues or questions. Thanks!