realsenseai/librealsense #5420: Question: surface area by calculating the size of each pixel

maxjundo, 2019-12-10:
| Camera Model                       |  {D415} | 
| Firmware Version                   | {05.11.15.00} | 
| Operating System & Version |  {Linux (Raspbian)}   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | {PRaspberry Pi 4}  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

I want to measure the area of ​​everything in depth frame (in the future I will take only the desired area using align).
 
It seems to me to calculate the area, I need to know the **distance for the pixel** and **pixel size** according to its distance to camera. Using the camera, I can get the distance to each pixel, but i don't know which size is that pixel.

Found a similar topic where the meaning is not clear: https://github.com/IntelRealSense/librealsense/issues/2730
I know how to convert pixel (x',y') into a 3D coordinate (x,y,z), but it’s not clear how this will help to calculate the area.

Maybe someone has experience in calculating the area? Advise please.

BadMachine, 2019-12-10:
To make it clear, are you exactly searching pixel size?
Pixel is not an area, it`s just a point in 2d space with some data like RGB, depth (for depth sensor), etc.
I don`t really understand what are you looking for.

maxjundo, 2019-12-11:
Thank you for your reply.

I’m looking for a way to calculate the surface area of ​​an object that fell into the depth camera zone, for e.g. guitar (invoice).

I assumed that I can calculate the area of ​​an object in three stages:
1) remove everything except the object from the frame
2) based on the size of the camera matrix, resolution and distance to the point, calculate the area of ​​a specific image pixel
3) Loop to recalculate the area of ​​all pixels in the image

Maybe this is wrong way, you can tell a way to calculate the area?

[image]

BadMachine, 2019-12-11:
@maxjundo
You can use setMouseCallback in opencv to pick the pixels.

Pseudo-code:
```c++
static void onMouse(int event,depth_frame, inrist, int x, int y, int, void*) {
	if (event != EVENT_LBUTTONDOWN) return;

        float pixel_distance_in_meters = depth.get_distance(x, y);
float planarPoint3d[3];
float pix[2] = {x,y};

              rs2_deproject_pixel_to_point(planarPoint3d, &inrist, pix, pixel_distance_in_meters);
...

	
}

...
find distance between A and B
...
```
https://answers.opencv.org/question/5915/how-to-use-cvsetmousecallback-in-a-c-class/

setMouseCallback("Find distance", onMouse, nullptr);

maxjundo, 2019-12-11:
@BadMachine
Sorry, but I'm don't understand how mouse handler will helps calulate the area of object. 

At the input, the function receives an image and depth. At the exit, I want to know the area of ​​the object in the image in square meters.

I realized that the pixel size will not help me, now I'm looking to the side Poisson surface reconstruction (creating mesh), and calculating area throught calculating area of polygons.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

> now I'm looking to the side Poisson surface reconstruction (creating mesh), and calculating area through calculating area of polygons

Thank you for the update. Will close this ticket unless there is any further question.

closed 2020-01-13 by RealSenseCustomerSupport