realsenseai/librealsense #5423: rs-enumerate-devices -s no longer provides a short summary of devices

phil-marble, 2019-12-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415| 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-55-generic | 
| Platform                                 | PC  |
| SDK Version                            | 2.29          | 
| Language                            |  N/A                        | 
| Segment			|  N/A                        | 

### Issue Description

`rs-enumerate-devices -s` no longer provides a short summary of devices. It still prints the summry first, but then goes on to print lots of detail too...

*incorrect* output in 2.29:
```
$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D415          81051206XXXX        05.08.15.00         
Intel RealSense D415          82121206XXXX        05.08.15.00         
Intel RealSense D415          80221206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
Device info: 
    Name                          :     Intel RealSense D415
    Serial Number                 :     XXXXX
    Firmware Version              :     05.08.15.00
    Recommended Firmware Version  :     05.11.15.00
    Physical Port                 :     /sys/devices/XXX
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD3
    Camera Locked                 :     YES
    Product Line                  :     D400
    Asic Serial Number            :     XXXXXXXXXXX
    Firmware Update Id            :     XXXXXXXXXXX

Device info: 
    Name                          :     Intel RealSense D415
    Serial Number                 :     XXXXX
    Firmware Version              :     05.08.15.00
    Recommended Firmware Version  :     05.11.15.00
    Physical Port                 :     /sys/devices/XXX
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD3
    Camera Locked                 :     YES
    Product Line                  :     D400
    Asic Serial Number            :     XXXXX
    Firmware Update Id            :     XXXXX

Device info: 
    Name                          :     Intel RealSense D415
<SNIP>
```

*correct* output in 2.15:

```
$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D415          81051206XXXX        05.08.15.00         
Intel RealSense D415          82121206XXXX        05.08.15.00         
Intel RealSense D415          80221206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
Intel RealSense D415          80951206XXXX        05.08.15.00         
```

dorodnic, 2019-12-11:
We'll check and fix. Short options is often convenient to have.
P.S. Please consider updating your firmware, 5.8.15 is super outdated

RealSenseCustomerSupport, 2020-02-27:
----------------------------------------------

Hi @ phil-marble, This issue is resolved by latest LRSV2.33. Could you please check it out? Thanks.

closed 2020-03-04 by dorodnic