realsenseai/librealsense #5424: Crashes Android app with null pointer dereference error

kafan1986, 2019-12-11:
Required Info |  
-- | --
Camera Model | { D415}
Firmware Version | {05.11.15.00}
Operating System & Version | {Android}
Platform | {Arm-Odroid}
SDK Version | { 2.30.0 }
Language | {C++/Java }
Segment | {others }

I have simple app where I am converting the RGB feed at 480px15 fps to my own queue of OpenCV mat and from a separate thread I use the display the mat by converting it to bitmap and then displaying it on an imageview in a loop.

I was testing the stability of android related to another issue https://github.com/IntelRealSense/librealsense/issues/5418 and received a crash on my app after few hours with 
```
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
Cause: null pointer dereference
```
The tombstone logs are provided below, I am pasting small portion of the tombstone below:

```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'ODROID/odroidn2/odroidn2:9/PQ3A.190801.002/20191204:eng/test-keys'
Revision: '0'
ABI: 'arm64'
pid: 8281, tid: 8612, name: Thread-3  >>> com.synapticsense.retailapp <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
Cause: null pointer dereference
    x0  0000000000000000  x1  00000072db037f10  x2  00000072e0400000  x3  00000072e048c860
    x4  0000000000000000  x5  0000000000000000  x6  00000072d8fa21a8  x7  0000000000000018
    x8  00000072ac8aa390  x9  f29e92054a92be5d  x10 0000000000000000  x11 0000000000000000
    x12 0000000000000001  x13 0000000000000010  x14 0000000000000000  x15 aaaaaaaaaaaaaaab
    x16 00000072acca5020  x17 00000072ac8eedf8  x18 0000000000000000  x19 00000072db037f10
    x20 00000072d2e26588  x21 00000072d2e24680  x22 00000072d2e26588  x23 0000000000000002
    x24 00000072dbd4fca0  x25 00000072ec980590  x26 0000000012c40138  x27 0000000000000064
    x28 0000000000000001  x29 00000072d2e246e0
    sp  00000072d2e24680  lr  00000072ac8a75a0  pc  00000072ac8a75ac

backtrace:
    #00 pc 00000000006725ac  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so (librealsense::frame::release()+184)
    #01 pc 00000000006b9e34  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so (rs2_release_frame+60)
    #02 pc 00000000005c7914  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so
    #03 pc 00000000005c32ac  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so (librealsense::processing_block::invoke(librealsense::frame_holder)+92)
    #04 pc 00000000006d2c18  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so (rs2_process_frame+76)
    #05 pc 0000000000649264  /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/lib/arm64/librealsense2.so (Java_com_intel_realsense_librealsense_ProcessingBlock_nInvoke+84)
    #06 pc 00000000000b5400  /dev/ashmem/dalvik-jit-code-cache (deleted) (com.intel.realsense.librealsense.ProcessingBlock.nInvoke+160)
    #07 pc 00000000000a1ac4  /dev/ashmem/dalvik-jit-code-cache (deleted) (com.intel.realsense.librealsense.Filter.process+84)
    #08 pc 00000000000aaa94  /dev/ashmem/dalvik-jit-code-cache (deleted) (com.intel.realsense.librealsense.FrameSet.applyFilter+68)
    #09 pc 00000000000b5968  /dev/ashmem/dalvik-jit-code-cache (deleted) (com.synapticsense.retailapp.VisionClasses.Streamer.stream+312)
    #10 pc 0000000000554cdc  /system/lib64/libart.so (art_quick_osr_stub+44)
    #11 pc 0000000000306f9c  /system/lib64/libart.so (art::jit::Jit::MaybeDoOnStackReplacement(art::Thread*, art::ArtMethod*, unsigned int, int, art::JValue*)+1996)
    #12 pc 000000000052ad5c  /system/lib64/libart.so (MterpMaybeDoOnStackReplacement+144)
    #13 pc 000000000054b9f0  /system/lib64/libart.so (ExecuteMterpImpl+33136)
    #14 pc 000000000005a108  /dev/ashmem/dalvik-classes2.dex extracted in memory from /data/app/com.synapticsense.retailapp-YN_Kgqu5-6Pd0YpwAg-Xxg==/base.apk!classes2.dex (deleted) (com.synapticsense.retailapp.VisionClasses.Streamer.stream+1076)
```

[tombstone_18.txt](https://github.com/IntelRealSense/librealsense/files/3947943/tombstone_18.txt)

The code to get frame is standard one and is taken from example app.
```
 try (FrameReleaser fr = new FrameReleaser()) {
                        FrameSet frames = mPipeline.waitForFrames(DEFAULT_TIMEOUT).releaseWith(fr);

                        FrameSet processedSet = frames.applyFilter(mDecimationFilter).releaseWith(fr)
                                .applyFilter(mTemporalFilter).releaseWith(fr)
                                .applyFilter(mHoleFillingFilter).releaseWith(fr);
```

MartyG-RealSense, 2020-03-05:
Apologies that you did not receive a response to your question.  Do you still need an answer please or can this case be closed?  Thanks!

kafan1986, 2020-03-05:
Although I never got the answer but it seems I avoided the error by restructuring certain part of the code but it is a shame, I never got to the bottom of the error. Anyways, I will close this issue out.

closed 2020-03-05 by kafan1986

MartyG-RealSense, 2020-03-05:
Thanks for your help!

joeyfedor, 2020-10-03:
@kafan1986 can you detail the steps you took to resolve this issue. I have the exact problem with an identical use case in opencv. I am interested in how you resolved this.

kafan1986, 2020-10-04:
@joeyfedor The issue lies with the release of frame. Look at the existing camera example code on android. You can make similar changes and it works OK.

joeyfedor, 2020-10-04:
@kafan1986  Are you stating that one must remove the frame releaser entirely to resolve this issue like the android capture example:

    try {
                try(FrameSet frames = mPipeline.waitForFrames()) {
                    try(FrameSet processed = frames.applyFilter(mColorizer)) {
                        mGLSurfaceView.upload(processed);
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }

kafan1986, 2020-10-04:
> @kafan1986 Are you stating that one must remove the frame releaser entirely to resolve this issue like the android capture example:
>
> ```
> try {
>             try(FrameSet frames = [...]

something like below has worked fine for me.

    try (FrameReleaser fr = new FrameReleaser()) {

                    FrameSet frameSet = mPipeline.waitForFrames(3000).releaseWith(fr);
             
                    FrameSet processedSet = frameSet.applyFilter(mDecimationFilter).releaseWith(fr)
                            .applyFilter(mTemporalFilter).releaseWith(fr);
                    Frame depth = processedSet.first(StreamType.DEPTH).releaseWith(fr);
                    DepthFrame depthFrame = depth.as(Extension.DEPTH_FRAME);
    }

gloaguen, 2024-02-14:
> Although I never got the answer but it seems I avoided the error by restructuring certain part of the code but it is a shame, I never got to the bottom of the error. Anyways, I will close this issue out.

@kafan1986 can you share your solution. We have a similar problem, see https://github.com/IntelRealSense/librealsense/issues/8296#issuecomment-1944014208

kafan1986, 2024-02-14:
> > Although I never got the answer but it seems I avoided the error by restructuring certain part of the code but it is a shame, I never got to the bottom of the error. Anyways, I will close this issue out.
> 
> @kafan1986 can you share your solution. We have a similar problem, see [#8296 (comment)](https://github.com/IntelRealSense/librealsense/issues/8296#issuecomment-1944014208)

I don't have access to any realsense device right now and I haven't been tracking the SDK development over last few years. BTW ain't the realsense hardware discontinued? Anyways if you need the solution, let me know, I might be able to dig out some code from old repositories.

gloaguen, 2024-02-14:
@kafan1986 if you can find some code, that could maybe help us.