realsenseai/librealsense #5425: how to capture the video on python?

jjh930910, 2019-12-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 7,10   | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|                            | 

### Issue Description
<Describe your issue / question / feature request / etc..>

i want to save the video file for rgb camera so i used the opencv cv2.videocapture , cv2.videowriter 
but this has problem that codec isn't available.(it can be saved video file but can't be performed)  i used the codec that can be used in opencv " DIVX, XVID, MJPG, X264, WMV1, WMV2" but all of this fail  how to capture the video?
i referred this page main code "https://076923.github.io/posts/Python-opencv-20/"

MartyG-RealSense, 2019-12-11:
The link below has a Librealsense script for Python that uses videowriter to save RGB in .Avi format.

https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Will you be needing further help with this?

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-04-05 by RealSenseCustomerSupport

matbord, 2023-04-12:
Hello, did you have any update on how to use cv2.video capture to capture the video on Python? I'm using a T265 camera