realsenseai/librealsense #5428: How to reset the camera using the camera serial number?

Dontla, 2019-12-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
I have six cameras connected to a computer, and I want to call the hardware_reset () function to reset two specific cameras without affecting the other cameras. I'm guessing whether I can reset it based on the serial number of the camera. But I don't know which function to use. Can you help me?

dorodnic, 2019-12-11:
From memory should be something like:
```python
import pyrealsense2 as rs2
ctx = rs2.context()
list = ctx.query_devices()
for dev in list:
      serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
      # compare to desired SN
      dev.hardware_reset()
```

Dontla, 2019-12-12:
list = ctx.query_devices () will store all connected camera objects, but I only want to perform hardware_reset () on two of them, I do n’t know how to do it

Dontla, 2019-12-12:
Thanks for your answer, I know how to operate it

closed 2019-12-17 by dorodnic

hfdlwt, 2023-08-02:
> From memory should be something like:
> 
> ```python
> import pyrealsense2 as rs2
> ctx = rs2.context()
> list = ctx.query_devices()
> for dev in list:
>       serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
>       # compare to desired SN
>       dev.hardware_reset()
> ```

Hi @dorodnic  ， I am a newer，Could you tell where to use this code to reset camera ？

ggsd-ddup, 2023-08-02:
Thank you for your reply @MartyG-RealSense ，this code need to update to librealsense or realsense-ros ？

MartyG-RealSense, 2023-08-02:
The code above is for a librealsense application.  It is possible though to launch pyrealsense2 code from within ROS by using a **node script**.  Examples of ROS node scripts for Python are provided by the RealSense ROS wrappers at the links below.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_camera/scripts

**ROS2**
https://github.com/IntelRealSense/realsense-ros/tree/ros2-development/realsense2_camera/scripts

ggsd-ddup, 2023-08-03:
Thanks @MartyG-RealSense ！ I know how to operate , but it did not work well for T265 , by using pyrealsense2.contest() is not detect T265 device please tell me why , thanks again.

MartyG-RealSense, 2023-08-03:
I cannot provide advice for T265 unfortunately as that model is no longer supported.

ggsd-ddup, 2023-08-03:
OK , thanks @MartyG-RealSense , when I using this code to reset camera as it is running , this code do not work ,as the picture showing  : RuntimeError: Unable to open device interface
[image]

MartyG-RealSense, 2023-08-03:
I don't know if T265 supports hardware reset.  It is suggested at https://github.com/IntelRealSense/librealsense/issues/3397#issuecomment-469964581 that a simple stop and start of the pipeline could reset its tracking.