realsenseai/librealsense #5430: Announcement: Intel® RealSense™ LiDAR camera L515

dorodnic, 2019-12-11:
<p align="center">[image]</p>

[Learn more @ `www.intelrealsense.com/lidar-camera-l515`](https://www.intelrealsense.com/lidar-camera-l515/)

MartyG-RealSense, 2019-12-11:
Awesome!  I've been checking for news of this camera for 7 months.  It's a rush to finally see it out in the wild.   \O/

skohlbr, 2019-12-11:
Looks very promising for a wide range of applications! I take that "perfect for indoor use" means that in outdoor use cases, the laser signal might be affected IR from the sun and thus degrade performance?

MartyG-RealSense, 2019-12-11:
The information page that Dorodnic linked to has an FAQ section at the bottom.  It says about outdoor use:

"Infrared light from the sun can interfere with the performance of the device which can degrade the quality of the depth images when used outdoors.  As a power efficient LiDAR camera, L515 will perform best indoors or in controlled lighting conditions."

sam598, 2019-12-12:
Very awesome news. I am excited to see how the depthmap compares to the active stereo cameras. There are a few quick questions I have:

1) It lists USB 3.1 Gen1 as a minimum spec. Since that spec is sometimes used interchangeably with USB 3.0, will this device be compatible with all 3.0 ports? Like for example a Raspberry Pi 4?

2) Is multicamera synchronization done through the same type of connectors as the D400 cameras? Will this new camera have built in ESD protection? Or is that still something we will need to solder ourselves?

3) Would it be possible to release an example .bag of a capture from the L515?

JimmyFroost, 2019-12-12:
I'm looking forward to seeing actual uses of this "lidar" camera! This seems to be a short-range  uni-directionnal lidar, nothing like 'usual' lidars like Velodyne or Hokuyo, maybe I'm too much used to 3D lidar, sorry ; )
 This looks to me as an upgrade of depth cameras, using 1 camera+ 1 lidar instead of 2 cameras, right ?

dorodnic, 2019-12-12:
@MartyG-RealSense - thanks, you should expect 99% of SDK capabilities to be available out of the box. 

@sam598 -
> It lists USB 3.1 Gen1 as a minimum spec. Since that spec is sometimes used interchangeably with USB 3.0, will this device be compatible with all 3.0 ports? Like for example a Raspberry Pi 4?

As with rest of supported devices, we are trying to enable wide compatibility, but cannot validate on all possible combinations. PRi 4 seem to work fine with L515 using USB3

>Is multicamera synchronization done through the same type of connectors as the D400 cameras?

Not the same, but similar connector.

> Would it be possible to release an example .bag of a capture from the L515?

Yes, will be released in near future.

MartyG-RealSense, 2019-12-13:
A line near the bottom of Intel's newsroom press release states: "Always ready to use, the L515 retains its depth accuracy throughout its lifespan without the need for calibration".

This was given special mention by a number of tech news websites, so clearly they recognize how much easier it will make the lives of users!

closed 2019-12-31 by RealSenseCustomerSupport

harmvandeven, 2020-02-10:
@dorodnic I'm really looking forward to the camera's I've pre-ordered. I won't be bothering you with delivery expectations ;) But I do have 2 questions in advance.

1. I'm planning to install the camera's on a ceiling, quite far away from my server room/computer. I know the camera will use USB 3, are there any recommended extenders? Have you tested anything with long cables?
2. Do you by any chance know if you wil be collaboration with FRAMOS again to create an industrial casing? Like with the D435e (https://www.framos.com/en/framos-depth-camera-d435e-camera-only-22806)

Thanks in advance!

MartyG-RealSense, 2020-02-10:
I'm not Dorodnic but I will offer my own 2 cents of opinion.   :)

1.  My personal recommendation for a supplier of USB extending cables and equipment is Newnex, who produce industrial-grade products that can extend a RealSense camera up to 100 meters with fiber optic cable.  My assumption would be that one of their cables should work as well with L515 as it does with the 400 Series cameras.

https://www.newnex.com/realsense-3d-camera-connectivity.php

https://www.youtube.com/watch?v=GLQgR1jT04M

According to the L515 data sheet, the L515 uses a USB Type C connector like the 400 Series does.

I should emphasise that other suppliers of industrial grade USB cable are available.  My personal recommendation of Newnex is primarily based on the fact that they can demonstrate in the video above that their products work with RealSense.

2.  Third-party camera collaborations are not limited to FRAMOS.  LIPS Corp also produce the LIPSedge AE400, another ethernet-based industrial RealSense camera with its own set of unique features.

https://www.lips-hci.com/product-page/lipsedge-ae400-industrial-3d-camera

As long as a partnership can offer a tangible "return on investment" to Intel (e.g ordering a volume of 100,000+ components) then Intel are happy to work in partnership together with companies on RealSense products.  FRAMOS officially partnered with Intel in 2017, which led to the FRAMOS D435e.  Third-party camera partnerships do require a commitment from companies of a certain amount of tech support though, as the company is responsible for providing the tech support for those third-party products as they are not Intel products.

fehrudi, 2020-02-22:
@dorodnic 
> > Would it be possible to release an example .bag of a capture from the L515?
> 
> Yes, will be released in near future.

Can you please tell me when we can expect the example .bag?

Thanks!

YohanBaillot, 2020-05-18:
Hello, we are interested to create a 3d model of room using the L515 camera. Is the L515 camera supported by the RealSense Unity SDK 2.0? Is it compatible with all the OSes supported by the SDK? Does Intel provides an app or does the SDK has capability to extend a point cloud using several scans of the L515? If not is there an external app that does that and on what platform does it work on? Does the L515 return color as well as depth or just depth?

MartyG-RealSense, 2020-05-18:
Hi @YohanBaillot   L515 uses the RealSense SDK 2.0.  99% of the SDK's functions are supported for L515 and it therefore supports the same programming languages.  It is also very quick and easy to convert an existing project for 400 Series cameras for the L515 just by changing a few lines of code.

The L515 has the same RGB imager component as the D415 and D435 cameras, and it can analyse 23 million accurate depth points per second.

You can scan with the L515 using a single camera or multiple L515s.  Like the 400 Series, multiple L515 can also be hardware synced together.

I hope that the further information resources in the link below will be helpful to you.  Please feel free to ask any further questions that you have.

https://support.intelrealsense.com/hc/en-us/community/posts/360043969714

YohanBaillot, 2020-05-18:
> Hi @YohanBaillot L515 uses the RealSense SDK 2.0. 99% of the SDK's functions are supported for L515 and it therefore supports the same programming languages. It is also very quick and easy to [...]

Thank you. How about software app able to create a point cloud or meshed model using scan coming from the camera?

MartyG-RealSense, 2020-05-19:
The 3D mode of the RealSense Viewer that is part of RealSense SDK 2.0 should be able to generate a point cloud from the L515 that can be exported as a **.ply** format file.  That file can then be imported into 3D modelling software capable of converting the point cloud data into a solid mesh.

An example of free software that can do this conversion is MeshLab.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab

Other programs that should be able to do the point cloud to mesh conversion include Blender, Pyntcloud (point cloud processing software for the Python language) and major 3D modelling software such as Maya (by using a plugin to add ply compatibility).  Because you would be converting a pre-created file, compatibility between the 3D software and L515 should not be an issue.

Regarding commercial scanning software for L515, the room scanning YouTube video apparently uses **DotProduct Dot3D Pro** - a DotProduct representative confirms this in the video's comments section.

https://www.dotproduct3d.com/dot3dpro.html

nawara72, 2020-06-16:
Hi Just received my L515, but the realsense-viewer (v2.35.0) can't seem to detect it??
Thanks

brentjackson, 2020-06-16:
> Hi Just received my L515, but the realsense-viewer (v2.35.0) can't seem to detect it??
> Thanks

Also received mine today and I initially had similar issues, but got it working pretty quickly remembering:

- Use a USB 3.0 or greater port. (Needed for framerate throughput.)
- Ensure sufficient power to run the camera. (Didn't work on my Dell tablet except when connected through a powered USB hub. ... although this could have also been lack of a good USB-C USB-C cable).

Hope that helps.

nawara72, 2020-06-16:
Thanks Brent, I am using a USB 3.0 port and tried it on both my laptop and desktop , using a powered usb hub, but no luck

MartyG-RealSense, 2020-06-16:
Hi @nawara72   I was going to say that official support for the L515 began in the latest SDK 2.35.2 released last week, so you should please use that SDK version.  I see that you have already received that information from a RealSense team member in the case that you posted.  Please ask further questions there so that the team member can assist you.  Good luck!

nawara72, 2020-06-16:
I was using 2.35.0, great i will try the new version 2.35.2, thanks for all your help

MartyG-RealSense, 2020-06-16:
@nawara72   You are very welcome!   Also, today the L515 data sheet document has been updated and an L515 CAD model added.

https://github.com/IntelRealSense/librealsense/issues/6606

nawara72, 2020-06-16:
Really it is great all the quick and useful help and advice i always get here. Great efforts and highly appreciated