realsenseai/librealsense #5431: T265: Feed a "pose aware" algorithm with poses from a .bag file

neilyoung, 2019-12-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

For development, test  and debug I would like to feed an algorithm, which is now obtaining its poses directly from the camera via `frames = pipe.wait_for_frames()`from a file, also in order to have reproducible inputs. 

Is there any sample, which provides some starter code or can somebody please just spread some hints, how to achieve that? I already used the `rs-record` tool. The question is now, how to read back such a bag programmatically (in C and Python) and how to extract the pose stream.

TIA

neilyoung, 2019-12-13:
No idea? Is it possible at all? I would expect, it would be possible.

neilyoung, 2019-12-13:
Oh great, found this: https://github.com/IntelRealSense/librealsense/issues/1919

closed 2019-12-13 by neilyoung

bfulkers-i, 2019-12-16:
@neilyoung If I understand correctly I think you may be looking for `cfg.enable_device_from_file("path_to_input_file.bag");` to make a virtual T265 that replays data from a file. See also https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

neilyoung, 2019-12-16:
@bfulkers-i Yes, thanks. Found it and it works. Right now I'm looking for a way to speed it up, since the playback seems to run with speed x 1 only. I s there a way? See also here please https://github.com/IntelRealSense/librealsense/issues/5447