realsenseai/librealsense #5432: rosbag not record the tf entire tree

szx0112, 2019-12-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.31) | 
| Operating System & Version |   { Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.0.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 5.12 }                          | 
| Language                            |  {ROS }                          | 

### Issue Description
I am new to ROS and Realsense. I try to record the tf tree of two d435i cameras. I use static_transform_publisher to connect each one of them with the base_link. The tf tree during the real image capturing shown below. 
[tf_tree_real_capture](https://github.com/IntelRealSense/librealsense/files/3953112/frames.1.pdf)

I use rosbag to record the /tf and /tf_static and play it with use_sim_time as true. However, I notice the rosbag fail to record the entire tf tree. The sub-tree of camera2 is missing in the result. 
[tf_tree_rosbag](https://github.com/IntelRealSense/librealsense/files/3953133/frames.pdf)

Below is the launch file of two cameras
[two_d435i.launch.txt](https://github.com/IntelRealSense/librealsense/files/3953196/two_d435i.launch.txt)

Any comment is appreciated

closed 2019-12-12 by szx0112

eric-schleicher, 2019-12-12:
Are you splitting the bags in any way?   It's a known _behavior_ that is you auto-split/roll rosbags by size/time, etc, the static transforms dont auto-propogate into later (than when originally published) bags.

Mikor-mkr, 2021-05-21:
I am experiencing the same issue but I am using different sensors, one T265 & one D435, when they publish the topics everything seems fine but when they are recorded the tree misses some transformations, any ideas? I am not splitting rosbags.