realsenseai/librealsense #5433: Frame tearing

sjdrc, 2019-12-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.159-tegra                                         | 
| Platform                                 | Jetson TX2  |
| SDK Version                            |  2.28.1-5ubuntu5~1.gbpb2a8ce                          | 
| Language                            |  C++ }                          | 
| Segment			|  N/A                         | 

We have observed frame tearing approximately 50 pixels from the top of the top of the image when using librealsense2 provided by Intel. When using our own self compiled driver version 2.28.0 we do not see this issue.

On my desktop (Ubuntu 18.04) we have version 2.31.0-0~realsense0.1791. Are you going to continue to provide updates for 16.04 arm64, or should we be compiling ourselves?

ev-mp, 2019-12-12:
@sjdrc hello, can you share thestreaming configuration parameters and some snapshots with the artifact?  
What Cmake flags are you building the SDK locally with?

The Debian packages with v2.31.0 for Arm64 (JetsonTX-optimized) will be released shortly.

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

Hi @sjdrc,

Do you still need help with this question?

Thank you!

closed 2019-12-22 by dorodnic