realsenseai/librealsense #5434: https://github.com/IntelRealSense/librealsense/issues/2818

LiranAz, 2019-12-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (18.04.3)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0                          | 
| Language                            |  Python                          | 
| Segment			|  other       | 

### Issue Description

Hi, 
I am extracting infrared frames from the SR300 camera and trying to get a higher resolution. 
when I use config.enable_stream(rs.stream.depth) it works fine (returning 640 by 480 for IR), but when I try to force any parameter (for example "config.enable_stream(rs.stream.infrared, width=320, height=240, format=rs.format.any, framerate=10)") I get an error saying "Couldn't resolve requests", even when requesting resolution identical or lower to the default ones.

I tried several different inputs that I found on the examples but none of them worked.
some of these inputs work fine for color, but not for IR.

is there something I can do to modify IR input from the camera?

Thanks!

MartyG-RealSense, 2019-12-12:
The SR300 only has the 640x480 IR channel, as its technology dates back to the start of 2016.  The error message that you received indicates that the camera was being asked to perform a task that it did not have support for.

[image]

LiranAz, 2019-12-12:
I understand, thanks!
but it does not work with these exact parameters (if I try to change the frame rate). could there be another issue?
can you send me a link to full table you attached please?

Thanks again!

MartyG-RealSense, 2019-12-12:
The table was taken from page 24 of the most recent edition of the data sheet document for the SR300 Product Family (SR300 and SR305).

https://www.intelrealsense.com/wp-content/uploads/2019/07/RealSense_SR30x_Product_Datasheet_Rev_002.pdf

MartyG-RealSense, 2019-12-12:
Below is an example of Python formatting of an infrared stream instruction for a 640x480 stream at 30 FPS.
rs_config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)

LiranAz, 2019-12-12:
It works, thanks.

MartyG-RealSense, 2019-12-12:
Awesome, glad I could help!   :)

closed 2019-12-16 by dorodnic