realsenseai/librealsense #5437: unable to setup on kernel 5.2

gbr1, 2019-12-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Ubuntu 18.04.3 | 
| Kernel Version (Linux Only)    |  5.2                                 | 
| Platform                                 | PC  |
| Segment			|  Robot/VR                          | 

### Issue Description
I'm unable to install from source librealsense2.
I think is something related to patch_realsense_ubuntu_lts.sh at line 86.
Looking into patch_utils.sh kernel 5.2 seems supported.
My kernel is `5.2.0-050200rc6-generic` and I can't downgrade.

Error given is:
```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-Ubuntu-hwe-edge-5.2.0-9.10_18.04.1 folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.
```

Thank you in advance.

ev-mp, 2019-12-12:
@gb1, we maintain the kernel patches for Ubuntu LTS line currently up to 5.0 with 5.3 scheduled next. Kernel v5.2 is not on Ubuntu LTS list, and will not be officially supported. 
We can provide some technical assistance in case you do decide to port the patches, but I would also recommend to review the requirements and to see whether you'll really benefit from this as with kernel 5.+ you already have the HW timestamps available.

In case you must have all the extra metadata attributes but the kernel patch path is not feasible then you can switch to the user-space UVC implementation provided by SDK by setting the `-DFORCE_RSUSB_BACKEND=true` cmake flag

closed 2019-12-17 by dorodnic

b4be1, 2020-07-08:
I am trying to apply `patch-realsense-ubuntu-lts.sh` on Linux ubuntu 5.3.0-1028-raspi2 #30~18.04.2-Ubuntu. However, I am getting the same error as OP: `fatal: You need to specify a tag name.` As I understood from other threads, kernel 5.3 is already supported (e.g., https://github.com/IntelRealSense/librealsense/issues/5748, https://github.com/IntelRealSense/librealsense/issues/5942). Can it be that the patching scripts need to be updated? Or can you give a suggestion how to build librealsense on Raspberry Pi 4 running Ubuntu 18.04?

b4be1, 2020-07-08:
The problem seems to be that there is no tag corresponding to my version `5.3.0-1028-raspi2` in the kernel repo https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/bionic/refs/. The closest is `Ubuntu-raspi2-5.3-5.3.0-1025.27~18.04.1`. So, the `grep` in the line https://github.com/IntelRealSense/librealsense/blob/2c1fc5444b8dfc84e0705a65333214792b6e5c1e/scripts/patch-realsense-ubuntu-lts.sh#L86 does not find a matching tag.

BellosC, 2021-03-18:
So how can we overcome this problem? 
(i have the same problem and i use:
-- raspberry pi 4 
-- 5.4.0-1030-raspi 
-- ubuntu 18.04)