realsenseai/librealsense #5438: Save frame to image file color(.png) and depth(.tiff) Android

vshalkov, 2019-12-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.2                                         | 
| Platform                                 | Android  |
| SDK Version                            |  RealSense SDK for Android OS                          | 
| Language                            |  C, opencv, kotlin, java }                          | 
| Segment			|  Smartphone }                          | 

### Issue Description
Hello, I need your help.
I use RealSense SDK for Android OS https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

I want to receive data from a frame (FrameReleaser()) and save it in a picture file
I have this code Config:
```
Config().use { config ->
         config.enableStream(StreamType.DEPTH, 640, 480, StreamFormat.Z16)
         config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGB8)
         pipeline.start(config)
     }
````
And i have this code:
```
   FrameReleaser().use { fr ->
             val frames = pipeline.waitForFrames(1000).releaseWith(fr)

             val depth = frames.first(StreamType.DEPTH).releaseWith(fr)
             val depthFrame: DepthFrame = depth.`as`(Extension.DEPTH_FRAME)
             depthFrame.releaseWith(fr)
             val byteArrayDepthFrame = ByteArray(depthFrame.width * depthFrame.height)
             depthFrame.getData(byteArrayDepthFrame)
             //TODO save frame to file image .png

             val colour = frames.first(StreamType.COLOR).releaseWith(fr)
             val colourFrame: VideoFrame = colour.`as`(Extension.VIDEO_FRAME)
             colourFrame.releaseWith(fr)
             val byteArrayColourFrame = ByteArray(colourFrame.width * colourFrame.height)
             colourFrame.getData(byteArrayColourFrame)
             //TODO save frame to file image .tiff

             upload(frames)
          }
```
I do not know how to convert bytes to the format I need, please help

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@vshalkov There's snapshot function in RealSenseViewer which can save color and depth image as *.png. Please refer to see if helps.
https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer
https://github.com/IntelRealSense/librealsense/blob/master/third-party/stb_image_write.h

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@vshalkov Any other questions? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-27:
----------------------------------------------

@vshalkov Will close this if no other questions. Thanks!

closed 2020-02-27 by RealSenseCustomerSupport