realsenseai/librealsense #5439: T265 Latency issue

xun-coder, 2019-12-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am having a noticeable delay using my T265 and AR glass, similar to Northstar setup...
also I noticed this statement in the know release note:
"Latency of 100ms (DSO-7745) - Will be fixed in a later FW release"
and this is not statement following on whether it's resolved or not.

bfulkers-i, 2019-12-12:
@xun-coder You should not see noticeable latency on T265, certainly not on the order of 100ms. For AR, we strongly recommend using pose prediction to use device velocity and acceleration to predict to the exact moment of rendering. You can find a simple example here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-predict

or a more full featured example here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

The release note you mention is about D400 and seems to have been fixed since firmware `5.9.3.0` (current firmware for D400 is `5.12.1.0`).

dorodnic, 2019-12-16:
Hi @xun-coder 
Sometimes T265 application generates perceived latency when logging to console is enabled in the background or while recording. 
Also, please make sure the SDK is built with flag `-DCMAKE_BUILD_TYPE=Release`.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Will you be needing any further assistance with this?

closed 2020-02-03 by RealSenseCustomerSupport