realsenseai/librealsense #5442: Publish the MacOS wheel to PyPI

TheButlah, 2019-12-12:
### Issue Description
<Describe your issue / question / feature request / etc..>

The official docs state that MacOS is supported and every release published to GitHub lists MacOS as a supported platform. However, only ubuntu and windows wheels are published to PyPI (Python Package Index), as seen here: https://pypi.org/project/pyrealsense2/#files

**Please publish the MacOS wheels to PyPI**. It makes it work out of the box with the rest of the python ecosystem's tooling. Having to compile using cmake is fine for manual set up and install, but is a real pain or impossible to incorporate into existing python tooling.

dorodnic, 2019-12-16:
Yes, we should. It is just a headache to automate as part of our CI release process, and I'm hesitant to do it manually from someones personal mac... I'll try to prioritize

TheButlah, 2020-02-10:
Any chance that this has been worked on since then? It's biting me in the butt once more :(

dorodnic, 2020-03-03:
@TheButlah - 
I just pushed latest release for python3.6 Mac OS
If you decide to give it a try we would appreciate your feedback

xiaoanne, 2020-03-09:
@dorodnic  is it possible to publish for python3.7 Mac OS? Thanks

TheButlah, 2020-04-03:
@dorodnic I don't see it listed on PyPi and pip installing `pyrealsense2` doesn't seem to work.

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@TheButlah Have submitted your requirement to our engineering team. Any update will let you know. Thanks!